KinesisCorporation/Adv360-Pro-ZMK #7: F13-F24 not working on MacOS.

joshvasquez, 2022-09-14:
# Expected Behavior

A key assigned as `&kp F13` in `adv360.keymap` should output `F13` and each subsequent function key up to and including `F24` should behave the same way on MacOS. These are supported by both MacOS and ZMK. 

# Current Behavior

The current behavior is that these keys produce no output whatsoever - It is not even registering as any keypress in MacOS. These keys are recognized by the operating system on other keyboards, including my modified Kinesis Advantage with QMK support. I'm not sure if these same keys produce any output on Windows or Linux as I don't currently have access to a running machine with those on it, though they may be affected as well. None of these register a keypress at all in MacOS. 

I personally use at least some of these function keys extremely regularly. For example, I use `F17` as a keybinding to toggle Alfred (a spotlight alternative) and `F19` to toggle between two different language input modes. 

# Steps to Reproduce

1. Assign any key on any layer to a Function key between F13-F24. According to ZMK's docs, the syntax should be written as`&kp F13`[^1] in the `adv360.keymap`  file for normal keypress. My current function layer keymap is:

```C
layer_fn {
            bindings = <
 &kp F1  &kp F2      &kp F3            &kp F4      &kp F5  &kp F6            &tog 1   &mo 3  &kp F7  &kp F8  &kp F9  &kp F10  &kp F11  &kp F12
 &kp F13 &kp F14     &kp F15           &kp F16     &kp F17 &kp F18           &none   &none   &kp F19 &kp F20 &kp F21 &kp F22  &kp F23  &kp F24
 &trans  &trans  &kp C_PREV  &kp C_PLAY_PAUSE  &kp C_NEXT  &trans             &none  &trans  &trans                   &trans   &trans    &none  &kp LEFT &kp DOWN &kp UP  &kp RIGHT &C_VOL_UP &trans
 &trans  &trans      &trans            &trans      &trans  &trans             &none   &none  &trans   &none   &none   &trans    &none    &none  &trans   &trans   &trans  &trans    &C_VOL_DN &trans
 &trans  &trans      &trans            &trans      &trans  &trans            &trans  &trans  &trans  &trans  &trans   &trans   &trans   &trans  &trans   &trans
            >;
```
Everything except for the F13-F24 functions as expected and there is no errors on build using this keymap. 

2. Build and Flash firmware onto device.
3. Attempt to type the function key. An application like Karabiner's event viewer [^2] allows you to view input events on MacOS. Alternatively, You can attempt to bind the key to something in the keyboard shortcuts panel in System Preferences. With a function F13-F24 key, it will correctly set the keybinding for the keyboard shortcut. As it stands, it continues waiting for a valid keypress. 

[^1]: https://zmk.dev/docs/codes#f-keys - one can intuit from the default keybindings that it should be possible to assign F13-F24 the same exact way as F1-F12 (which work fine), hence `&kp F13`.

[^2]: https://karabiner-elements.pqrs.org/ - I've bypassed karabiner entirely so it should not be causing the issue.

KinesisCorporation, 2022-09-20:
Looks like there is a conflict with NKRO at the moment. Commenting out that config line makes F13-F24 work again. We are investigating a fix. Sorry!

JW9506, 2022-09-23:
I can reproduce on Windows.

JW9506, 2022-09-23:
> Looks like there is a conflict with NKRO at the moment. Commenting out that config line makes F13-F24 work again. We are investigating a fix. Sorry!

I have just commented out `CONFIG_ZMK_HID_REPORT_TYPE_NKRO` but ALL keys would stop working.

KinesisCorporation, 2022-09-23:
thanks for the update.

joshvasquez, 2022-09-24:
> I have just commented out `CONFIG_ZMK_HID_REPORT_TYPE_NKRO` but ALL keys would stop working.

I changed it to value `n` instead of commenting it out and it works, though presumably without NKRO: 

```
❯ rg NKRO
config/boards/arm/adv360/adv360_right_defconfig
52:CONFIG_ZMK_HID_REPORT_TYPE_NKRO=n

config/boards/arm/adv360/adv360_left_defconfig
53:CONFIG_ZMK_HID_REPORT_TYPE_NKRO=n
```

Acceptable workaround for me, but obviously doesn't help those that require NKRO.

JW9506, 2022-09-24:
> I changed it to value `n` instead of commenting it out and it works, though presumably without NKRO:
>
> ```
> ❯ rg NKRO
> config/boards/arm/adv360/adv360_right_defconfig [...]

I cannot reproduce this whether commenting those lines out OR setting them to `n`... 

I am using `Windows11Pro`.

joshvasquez, 2022-09-24:
Does your config still work when you revert it to the original? Seems odd that it'd only work when set to `y`.

JW9506, 2022-09-24:
> Does your config still work when you revert it to the original? Seems odd that it'd only work when set to `y`.

Yes - I immediately flash my previous firmware then all keys would start working again. setting n or commenting **all** keys would not register.

frankrei, 2022-10-05:
I too, have tried to set the value to =n put that did not work on my end. No `F13-F20`. Tested using Karabiner-EventViewer.

KinesisCorporation, 2022-10-05:
Hi there!

- Are you guys using the v2.0 branch?
- Did you modify both left and right deconfig files?
- Did you eliminate any spaces around the "=n"?

We've tested here several times and its working for us.

JW9506, 2022-10-05:
> Hi there!
> 
> * Are you guys using the v2.0 branch?
> * Did you modify both left and right deconfig files?
> * Did you eliminate any spaces around the "=n"?
> 
> We've tested here several times and its working for us.

Right now I am using v2.0 - NKRO=y `->` F13+ are not working.

For now I just think disabling NKRO is too much of a sacrifice even if it works.

frankrei, 2022-10-05:
Hi,

Thanks. I actually worked off the v1.0 branch, but I forgot. Once I updated
to the v2.0 branch and make the settings changes, F13 to F24 worked
correctly on the Mac.

Best regards,

Frank

ReFil, 2022-10-07:
Is anyone aware of other keyboards that support those keys with NKRO, This issue could be a fundamental limitation of the USB HID protocol, and a bit of rudimentary research hasn't brought up any keyboards that support both

lukebouch, 2023-01-04:
I ran into the same issue. Any permanent solutions?

frankrei, 2023-01-05:
Upgrading to the v2 branch activated the function keys but I had to switch off NKRO. No news since.

ReFil, 2023-01-05:
Turning off NKRO is the only option. Is this supported on other keyboards/firmwares as I cannot find examples of both NKRO and these keys working, I might not be looking in the right place

bjrnt, 2023-01-14:
I tried to find an explanation of what disabling NKRO for this keyboard means but came up short-handed. Could someone explain what value of KRO this keyboard reverts to if this config is disabled?

felixakiragreen, 2023-09-15:
Thank you, I had to disable NKRO it too for these to work. (I'm on branch v3)

@KinesisCorporation Is this something that can be fixed, so that NKRO and F13-24 will work again?

ReFil, 2023-10-21:
On the latest update (https://github.com/ReFil/zmk/commit/169aa900f4a9777df6f9d2a560eccd841d3444c7) this should be fixed. You might need to unpair and repair to the host computer.

ReFil, 2023-10-31:
This fix has been re-implemented in the beta branch. If any users are willing to test this and provide feedback it would be much appreciated. https://github.com/KinesisCorporation/Adv360-Pro-ZMK#beta-testing

ReFil, 2023-11-07:
This has now been changes to use a config option `CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT`, set to y for the full  NKRO range

closed 2023-11-08 by ReFil