KinesisCorporation/Adv360-Pro-ZMK #8: Build action failed

thinium, 2022-09-16:
Hi! I just tried to build the ZMK firmware for the first time. But the action failed in two places. Is it possible there is an issue in the build script? How can I fix them?

Here is my build action: https://github.com/thinium/Adv360-Pro-ZMK/actions/runs/3061013510/jobs/4940248396#step:8:37

[image]
[image]

thinium, 2022-09-16:
fixed.... it's cost by an empty biding

dustMason, 2022-09-18:
i'm having the same error - can you describe how you fixed it? i changed only a single key binding but am seeing this error on my github actions build

thinium, 2022-09-18:
@dustMason in my case there is a macro that is empty (it contains no keys). After removing that macro everything works fine. Yours seems to be a different issue from mine

dustMason, 2022-09-18:
Thank you – it turns out that my error was also caused by an empty macro. I rolled back my changes one by one until i discovered the empty macro. It's easy to make this mistake in the GUI editor – I suggest adding a validation in the editor to prevent saving an empty macro.

KinesisCorporation, 2022-09-19:
Appreciate all the dialogue guys!

1. We're about to publish an update that merges our custom features with the latest from ZMK.
2. And we're starting a troubleshooting tip with a list of common mistakes here:
https://kinesis-ergo.com/support/kb360pro/#troubleshooting