KinesisCorporation/Adv360-Pro-ZMK #9: Why is the Adv360-Pro-GUI Repository not Public?

steinbrueckri, 2022-09-16:
I ask because it is not possible to select "none" for a key in the Adv360-Pro-GUI. I love to contribute and create a PR but it's not possible. 😢

KinesisCorporation, 2022-09-16:
Just toggled to public. Appreciate your contributions!

steinbrueckri, 2022-09-16:
@KinesisCorporation Thanks 🙏

steinbrueckri, 2022-09-17:
@KinesisCorporation After some research, it makes sense why "Adv360-Pro-GUI" was not public because it is just the build from the Vue App in the Repository https://github.com/KinesisCorporation/Adv360-Pro-KeymapEditor. 

Correct me if I'm wrong but the KeyCodes are loaded from the Vue application from the [adv360-pro-api](https://adv360-pro-api.herokuapp.com/keycodes). So I guess I cannot fix the issue by myself. 😔

steinbrueckri, 2022-09-19:
THX for the clarification I was not aware of that. @Giljjj @KinesisCorporation

closed 2022-09-19 by steinbrueckri