KinesisCorporation/Adv360-Pro-ZMK #13: Tap Dance does not work - devicetree error: ... lacks binding

taylon, 2022-09-22:
Hello,

When trying to setup tap dance behaviors I get the following error:

```
devicetree error: binding controller <Node /behaviors/tap_dance_xxx 'adv360_left.dts.pre.tmp'> for <Node /keymap/default_layer in 'adv360_left.dts.pre.tmp'> lacks binding
```

Here is the keymap definition: https://github.com/taylon/Adv360-Pro-ZMK/blob/V1.0/config/adv360.keymap
Here is a build with the error: https://github.com/taylon/Adv360-Pro-ZMK/actions/runs/3102414540/jobs/5024689982

- I tried to update the ZMK docker image to `stable` and `3.0` instead of `2.4` and that did not make a difference.
- If I remove the reference to the tapdance behavior from the layer definition it compiles just fine.
- Seached in the zmk and zephyr repos to see if there is any reference to this error and could not find anything
- The error message seems to be triggered [here](https://github.com/zmkfirmware/zephyr/blob/v3.0.0%2Bzmk-fixes/scripts/dts/python-devicetree/src/devicetree/edtlib.py#L1368 ) 
- I tried to use the exact same tap dance definiton showed in the documentation [here](https://zmk.dev/docs/behaviors/tap-dance#basic-example-counter) (I mean I replaced the whole keymap with that) and got the same error, which is what makes me believe that this might be an issue with the Adv360 setup as opposed to a bug in zmk (I might be wrong though)

mpentecost, 2022-09-22:
@taylon have you tried building your firmware from the V2.0-beta branch?  I am running that branch locally and was able to successfully build your keymap.

taylon, 2022-09-23:
Oh nice, I did not notice that, it worked indeed.

Thanks,

closed 2022-09-23 by taylon