KinesisCorporation/Adv360-Pro-ZMK #16: Update bin/build.sh to support local builds

zemackdaddy, 2022-09-25:
I noticed that you deleted lines in .github/workflows/build.yml but not in bin/build.sh.

I cannot build locally (mac) without removing the matching lines in bin/build.sh.

Doesn't it make sense to remove it in that file as well?

`#!/usr/bin/env bash

set -e

PWD=$(pwd)
TIMESTAMP=$(date -u +"%Y%m%d%H%M%S")

\# West Build (left)
west build -s zmk/app -d build/left -b adv360_left -- -DZMK_CONFIG="${PWD}/config"
\# Adv360 Left DTS File
\# remove this cat -n build/left/zephyr/adv360_left.dts.pre.tmp
\# Adv360 Left Kconfig file
cat build/left/zephyr/.config | grep -v "^#" | grep -v "^$"
\# West Build (right)
west build -s zmk/app -d build/right -b adv360_right -- -DZMK_CONFIG="${PWD}/config"
\# Adv360 Right DTS File
\# remove this cat -n build/right/zephyr/adv360_right.dts.pre.tmp
\# Adv360 Right Kconfig file
cat build/right/zephyr/.config | grep -v "^#" | grep -v "^$"
\# Rename zmk.uf2
cp build/left/zephyr/zmk.uf2 ./firmware/${TIMESTAMP}-left.uf2 && cp build/right/zephyr/zmk.uf2 ./firmware/${TIMESTAMP}-right.uf2
`

ReFil, 2022-09-25:
Which branch are you using?

zemackdaddy, 2022-09-25:
I am sorry I conflated two things here.  This issue should be in a different repo, referencing this new commit.

zemackdaddy, 2022-09-25:
[V2-beta](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/tree/V2.0-beta)

This references the paths you removed in [.github/workflows/build.yml](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V2.0-beta/bin/build.sh)

zemackdaddy, 2022-09-25:
updated my original comment above.  Apologies.  This is all a bit new.

zemackdaddy, 2022-09-25:
[My repo](https://github.com/zemackdaddy/Adv360-Pro-ZMK/tree/V2.0-beta)

For context, I split setup.sh into two files.  I also split Dockerfile. 
I modified the process so that I run a setup–zmk.sh to prepare the base image.
I then run setup-build.sh followed by run-local.sh to pull in my local changes and build locally.

The important part is that run-local.sh if I don't comment out the same cat statements you removed in the yaml file [here](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V2.0-beta/.github/workflows/build.yml).

My local build will fail in the same manner if I just run setup.sh and run.sh.

So since you modified a file that removes the cat statements, I thought maybe they should also be removed in the build script?

ReFil, 2022-09-25:
Ah I understand now. Yes, that makes sense now. My apologies I didn't do the local building stuff. Will make those fixes asap, thank you for bringing this to my attention

zemackdaddy, 2022-09-25:
I'm glad I saw your changes.  I was stumped!

ReFil, 2022-10-07:
Closing as has been fixed

closed 2022-10-07 by ReFil