KinesisCorporation/Adv360-Pro-ZMK #18: Vendor and Product ID are 0

joshfrench, 2022-09-27:
I was trying to configure [Karabiner](https://karabiner-elements.pqrs.org/) to ignore certain mods that can now be programmed directly on the Advantage, while leaving them enabled on my built-in keyboard. Karabiner does this by checking the vendor and/or product ID of the device.

According to Karabiner these values are missing or zeroed out:
```
[
    {
        "device_id": 4294970112,
        "is_karabiner_virtual_hid_device": false,
        "is_keyboard": true,
        "is_pointing_device": false,
        "location_id": 2150578859,
        "manufacturer": "Manufacturer",
        "product": "Adv360 Pro",
        "product_id": 0,
        "transport": "Bluetooth Low Energy",
        "vendor_id": 0
    },
    {
        "device_id": 4294968566,
        "is_built_in_keyboard": true,
        "is_karabiner_virtual_hid_device": false,
        "is_keyboard": true,
        "is_pointing_device": false,
        "location_id": 18446744071567310848,
        "manufacturer": "Apple Inc.",
        "product": "Apple Internal Keyboard / Trackpad",
        "product_id": 638,
        "transport": "USB",
        "vendor_id": 1452
    },
    {
        "device_id": 4294968517,
        "is_karabiner_virtual_hid_device": false,
        "is_keyboard": true,
        "is_pointing_device": true,
        "location_id": 3982802855,
        "manufacturer": "Logitech",
        "product": "M720 Triathlon",
        "product_id": 45077,
        "transport": "Bluetooth Low Energy",
        "vendor_id": 1133
    }
] 
```

The zmk docs indicate the defaults are non-zero, so I'm a little surprised these are just plain missing:

Config | Type | Description | Default
-- | -- | -- | --
CONFIG_USB_DEVICE_VID | int | The vendor ID advertised to USB | 0x1D50
CONFIG_USB_DEVICE_PID | int | The product ID advertised to USB | 0x615E

There are workarounds so this isn't critical, but it would be nice to have known identifiers to rely on. Thanks!

ReFil, 2022-10-07:
Not sure why they were set to zero, definitely shouldn't be happening. I've forcibly written them into the config now, so if you sync fork you should be sorted now

joshfrench, 2022-10-10:
Looks like this may also require a fix on the zmk side: https://github.com/pqrs-org/Karabiner-Elements/issues/3236

Thanks for adding these though, this should make it easier to verify once the upstream fix has been merged.

closed 2022-10-10 by joshfrench

ReFil, 2022-10-10:
These fixes have been merged into zmk main, I can update the adv360 pro repo to implement them too

joshfrench, 2022-10-11:
Success! I did have to apply the change from #25, and the manufacturer string still needs updating:

```
    {
        "device_id": 4294989613,
        "is_karabiner_virtual_hid_device": false,
        "is_keyboard": true,
        "is_pointing_device": false,
        "location_id": 2150578859,
        "manufacturer": "ZMK Project",
        "product": "Adv360 Pro",
        "product_id": 24926,
        "transport": "Bluetooth Low Energy",
        "vendor_id": 7504
    }
```

But here's a PR for that: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/pull/28

Thanks for this change :)