KinesisCorporation/Adv360-Pro-ZMK #19: Problem with long macros

skamelone, 2022-09-28:
I have some issues with long macros, the first letter is typed multiple times. That doesn't happen with macros which are less than ten characters long.

I read in the ZMK documentation [here](https://zmk.dev/docs/behaviors/macros#behavior-queue-limit) that it's recommended to set `CONFIG_ZMK_BEHAVIORS_QUEUE_SIZE=512` for long macros. I defined it in `adv360.yaml` but got the following error:

```
warning: attempt to assign the value '512' to the undefined symbol ZMK_BEHAVIORS_QUEUE_SIZE
```

Another thing that doesn't work is to set `wait-ms` and `tap-ms` in the macros itself
```
wait-ms = <30>;
tap-ms = <40>;
```
I'm getting an error the error:
```
'tap-ms' appears in /behaviors/macro_test_email in adv360_left.dts.pre.tmp, but is not declared in 'properties:' in /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/dts/bindings/behaviors/zmk,behavior-macro.yaml
```

Any help would be appreciated.

KinesisCorporation, 2022-09-28:
Which branch of firwmare are you running?

skamelone, 2022-09-28:
branch out of `v1`

KinesisCorporation, 2022-09-28:
Please upgrade to v2 and try again. Wait and tap may even not be supported in v1. Thanks!

skamelone, 2022-09-29:
That worked, thank you

closed 2022-09-29 by skamelone

KinesisCorporation, 2022-09-29:
yay!