KinesisCorporation/Adv360-Pro-ZMK #20: Previously working builds fail on West Build (right)

dr-neptune, 2022-10-01:
I'm using the V-2.0 beta branch. Initially I thought I broke the config, but I'm getting the same error when I try to build the default keymap config and when I rerun previously successful builds:

> FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/right/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/right/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/usb/device/libdrivers__usb__device.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/led/libdrivers__led.a  zephyr/drivers/led_strip/libdrivers__led_strip.a  zephyr/drivers/pwm/libdrivers__pwm.a  zephyr/drivers/spi/libdrivers__spi.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/drivers/kscan/libzmk__drivers__kscan.a  modules/drivers/sensor/battery/lib..__zmk__app__drivers__sensor__battery.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp"  -L/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/right/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/right/zephyr && /usr/local/lib/python3.8/dist-packages/cmake/data/bin/cmake -E echo

JW9506, 2022-10-01:
Same here. I have been getting the same error while trying to build the latest official commit in 2.0Beta.

ReFil, 2022-10-02:
Apologies, should be fixed now, can rerun affected jobs in the github action summary screen when you click on the failed action

dr-neptune, 2022-10-02:
Build rerun was successful! Thanks for the quick turnaround -- I'm excited to get back to tinkering with my zmk config

closed 2022-10-02 by dr-neptune

ai212983, 2022-10-04:
Would be nice if you guys linked commits what and where exactly was changed.

JW9506, 2022-10-04:
> Would be nice if you guys linked commits what and where exactly was changed.

Maybe it is something upstream and it is nontrial for us to understand 😄

ai212983, 2022-10-04:
@JW9506 most likely it is [this commit](https://github.com/ReFil/zmk/commit/a981282fb57e87d389f80f15c45d1f4c309c486d#diff-c9f41eec2bc3633fa65b33fca41f94419f473db5180091ad313892d1a62c4ef4), which is misleadingly named "battery life improvements".

ReFil, 2022-10-04:
That commit does bring battery life improvements. It's a refactor to change how the brightness of the LEDs is controlled. The description is not misleading

gohanlon, 2022-10-04:
@ReFil Ah, was that not the commit that fixes the build, then?

ReFil, 2022-10-04:
To maintain a clean history the commit was amended to fix the bug

ai212983, 2022-10-04:
@ReFil This commit does two things

- fix broken build 
- improve battery life. 

Commit message informs of minor improvement and does not informs about fixing showstopper. It's totally misleading and counter-intuitive to me. YMMV, of course.

ReFil, 2022-10-04:
It didn't fix the broken build. The build was working before that commit. The building hasn't been broken for 21 days (time since the prior commit) 

A bug was introduced in the commit and in order to maintain a clean commit history I amended the commit and force pushed. The description of the commit is accurate for what the commit does

ai212983, 2022-10-04:
Well, this commit actually DID fix the broken build. Which was broken by another commit, removed by your history rewrite. 

> A bug was introduced in the commit and in order to maintain a clean commit history I amended the commit and force pushed. 

Amending does not just alter the most recent commit, it replaces it entirely, meaning the amended commit will be a new entity with its own ref. 

There are _Strong Opinions_ about force pushing and history rewriting, and those opinions are not totally baseless.