KinesisCorporation/Adv360-Pro-ZMK #26: setup.sh fails in V2.0 branch (macOS)

humblehacker, 2022-10-10:
I've successfully used the V1.0 branch to configure my adv360, but when I checkout a clean V2.0 branch and run `./setup.sh`, it fails. I've completely wiped Docker using the "Clean/Purge data" option from the Docker Desktop troubleshooting pane before running setup. Any ideas?

```
~/src/Adv360-Pro-ZMK (V2.0) > ./setup.sh
[+] Building 90.8s (11/14)
 => [internal] load build definition from Dockerfile                                                                                                                                                                                                                                                                     0.0s
 => => transferring dockerfile: 331B                                                                                                                                                                                                                                                                                     0.0s
 => [internal] load .dockerignore                                                                                                                                                                                                                                                                                        0.0s
 => => transferring context: 2B                                                                                                                                                                                                                                                                                          0.0s
 => [internal] load metadata for docker.io/zmkfirmware/zmk-build-arm:2.4                                                                                                                                                                                                                                                 1.9s
 => [auth] zmkfirmware/zmk-build-arm:pull token for registry-1.docker.io                                                                                                                                                                                                                                                 0.0s
 => [1/9] FROM docker.io/zmkfirmware/zmk-build-arm:2.4@sha256:5985a5a70be4262d6fc39aa53dd0551d57b856d3010be5db4f2ba56733e4f905                                                                                                                                                                                          32.4s
 => => resolve docker.io/zmkfirmware/zmk-build-arm:2.4@sha256:5985a5a70be4262d6fc39aa53dd0551d57b856d3010be5db4f2ba56733e4f905                                                                                                                                                                                           0.0s
 => => sha256:5985a5a70be4262d6fc39aa53dd0551d57b856d3010be5db4f2ba56733e4f905 955B / 955B                                                                                                                                                                                                                               0.0s
 => => sha256:a5213ffd6da02377d71d48b23da7b2be47b893bc0fe8afd59357572da48981d3 4.22kB / 4.22kB                                                                                                                                                                                                                           0.0s
 => => sha256:6e640006d1cd2f4db90bc3010afc398dfd9da7bb7d287bbf7f89088f6252407b 27.11MB / 27.11MB                                                                                                                                                                                                                        16.6s
 => => sha256:8dbc34bfbef0018fe9049b7eb81e17cb7ca8851b84269c46bffcf6ed839c97c8 142.36MB / 142.36MB                                                                                                                                                                                                                      24.0s
 => => sha256:ebc4963c7ce64184d1f8c6a972447a1a7bd747b1e669ecc64ad102adcbdee9e1 190.40MB / 190.40MB                                                                                                                                                                                                                      24.9s
 => => extracting sha256:6e640006d1cd2f4db90bc3010afc398dfd9da7bb7d287bbf7f89088f6252407b                                                                                                                                                                                                                                0.8s
 => => extracting sha256:8dbc34bfbef0018fe9049b7eb81e17cb7ca8851b84269c46bffcf6ed839c97c8                                                                                                                                                                                                                                3.3s
 => => extracting sha256:ebc4963c7ce64184d1f8c6a972447a1a7bd747b1e669ecc64ad102adcbdee9e1                                                                                                                                                                                                                                4.7s
 => [internal] load build context                                                                                                                                                                                                                                                                                        0.0s
 => => transferring context: 31.13kB                                                                                                                                                                                                                                                                                     0.0s
 => [2/9] RUN mkdir -p /app/firmware                                                                                                                                                                                                                                                                                     0.4s
 => [3/9] WORKDIR /app                                                                                                                                                                                                                                                                                                   0.0s
 => [4/9] COPY config/west.yml config/west.yml                                                                                                                                                                                                                                                                           0.0s
 => [5/9] RUN west init -l config                                                                                                                                                                                                                                                                                        0.9s
 => ERROR [6/9] RUN west update                                                                                                                                                                                                                                                                                         55.2s
------
 > [6/9] RUN west update:
#11 0.963 === updating zmk (zmk):
#11 0.964 --- zmk: initializing
#11 0.994 Initialized empty Git repository in /app/zmk/.git/
#11 1.054 --- zmk: fetching, need revision adv360-z3
#11 4.632 From https://github.com/refil/zmk
#11 4.632  * branch            adv360-z3  -> FETCH_HEAD
#11 4.993 HEAD is now at b3322a5e Change default settings
#11 5.340 HEAD is now at b3322a5e Change default settings
#11 5.551 === updating zephyr (zephyr):
#11 5.551 --- zephyr: initializing
#11 5.581 Initialized empty Git repository in /app/zephyr/.git/
#11 5.641 --- zephyr: fetching, need revision v3.0.0+zmk-fixes with --depth 1
#11 50.95 From https://github.com/zmkfirmware/zephyr
#11 50.95  * branch                v3.0.0+zmk-fixes -> FETCH_HEAD
#11 50.96  * [new tag]             v1.0.0         -> v1.0.0
[... cut ...]
#11 54.93  - Key 'groups' was not defined. Path: '/projects/38'.
#11 54.93  - Key 'groups' was not defined. Path: '/projects/42'.
#11 54.93  - Key 'groups' was not defined. Path: '/projects/43'.
#11 54.93  - Key 'groups' was not defined. Path: '/projects/44'.
#11 54.93  - Key 'groups' was not defined. Path: '/projects/45'.
#11 54.93  - Key 'groups' was not defined. Path: '/projects/46'.
#11 54.93  - Key 'group-filter' was not defined. Path: ''.
------
executor failed running [/bin/sh -c west update]: exit code: 1
```

humblehacker, 2022-10-11:
PR #25 fixes this issue.

ReFil, 2022-10-23:
Closed as pr merged and issue fixed

closed 2022-10-23 by ReFil