KinesisCorporation/Adv360-Pro-ZMK #34: Bluetooth only connects to one Operating System on dual-boot systems

Schilt, 2022-10-21:
I have dual boot systems with Windows and Ubuntu GNU/Linux.  The keyboard using bluetooth will only connect to one operating system on an individual PC.  It won't connect to both.  My Logitech MX Anywhere mouse though does connect to both systems.  I am only using one Operating System at a time, of course.  I am using the multiple device connections feature of the keyboard.  Other than this it works well!  Thanks!

abenz1267, 2022-10-22:
see here: https://github.com/zmkfirmware/zmk/issues/1503

ReFil, 2022-10-23:
As the other issue mentions this is because ZMK is strict with bluetooth security and doesn't allow different keys for the same physical device. There's a workaround for the time being. I do not plan on changing the bluetooth security behaviour until the recquisite changes are implemented in ZMK main

closed 2022-10-23 by ReFil