KinesisCorporation/Adv360-Pro-ZMK #37: Build fails on macOS (m1)

gggdomi, 2022-10-25:
Hi,

I just tried to build the firmware on my M1 mac without success.
The steps I followed are minimal:

1. fresh docker install (with rosetta)
2. `git clone` this project
3. `make all`

Do you know what may be going wrong?

```
➜  Adv360-Pro-ZMK git:(V2.0) make all
docker build --tag zmk --file Dockerfile .
[+] Building 269.4s (12/12) FINISHED
 => [internal] load build definition from Dockerfile                                                                          0.0s
 => => transferring dockerfile: 287B                                                                                          0.0s
 => [internal] load .dockerignore                                                                                             0.0s
 => => transferring context: 2B                                                                                               0.0s
 => [internal] load metadata for docker.io/zmkfirmware/zmk-build-arm:stable                                                   1.6s
 => [1/7] FROM docker.io/zmkfirmware/zmk-build-arm:stable@sha256:acea7abae2fe324e574e33f63761b1c20bb60714fb528a7987b093dc32  24.8s
 => => resolve docker.io/zmkfirmware/zmk-build-arm:stable@sha256:acea7abae2fe324e574e33f63761b1c20bb60714fb528a7987b093dc322  0.0s
 => => sha256:4b6fd231874ac99e60078328dbb4508ced1ea25841107e84b65ceafef831fdaa 184.23MB / 184.23MB                           12.5s
 => => sha256:ec7db53eba75ff4bec0a8c9e8ea486f6625864606c7f25dfa2f472a27b3c1f7f 250.91MB / 250.91MB                           18.5s
 => => sha256:acea7abae2fe324e574e33f63761b1c20bb60714fb528a7987b093dc32206e06 955B / 955B                                    0.0s
 => => sha256:23cad88e0ef6f8f61c3f6534901330f9fd41f93f68e0c2cd63e3e2a71c739a15 4.65kB / 4.65kB                                0.0s
 => => sha256:08c01a0ec47e82ebe2bec112f373d160983a6d1e9e66627f66a3322bc403221b 28.56MB / 28.56MB                              2.3s
 => => extracting sha256:08c01a0ec47e82ebe2bec112f373d160983a6d1e9e66627f66a3322bc403221b                                     1.2s
 => => extracting sha256:4b6fd231874ac99e60078328dbb4508ced1ea25841107e84b65ceafef831fdaa                                     6.1s
 => => extracting sha256:ec7db53eba75ff4bec0a8c9e8ea486f6625864606c7f25dfa2f472a27b3c1f7f                                     5.9s
 => [internal] load build context                                                                                             0.0s
 => => transferring context: 1.04kB                                                                                           0.0s
 => [2/7] WORKDIR /app                                                                                                        0.3s
 => [3/7] COPY config/west.yml config/west.yml                                                                                0.0s
 => [4/7] RUN west init -l config                                                                                             0.8s
 => [5/7] RUN west update                                                                                                   233.2s
 => [6/7] RUN west zephyr-export                                                                                              3.6s
 => [7/7] COPY bin/build.sh ./                                                                                                0.0s
 => exporting to image                                                                                                        4.9s
 => => exporting layers                                                                                                       4.9s
 => => writing image sha256:fffb06c16a5a8ec4f077a4cf3db4bd0d3d15a6204079c605e2bb3758cf0cb55b                                  0.0s
 => => naming to docker.io/library/zmk                                                                                        0.0s

Use 'docker scan' to run Snyk tests against images to find vulnerabilities and learn how to fix them
docker run --rm -it --name zmk \
		-v /Users/gui/Dev/Adv360-Pro-ZMK/firmware:/app/firmware \
		-v /Users/gui/Dev/Adv360-Pro-ZMK/config:/app/config:ro \
		-e TIMESTAMP=20221025045710 \
		zmk
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
-- west build: generating a build system
Including boilerplate (Zephyr base): /app/zephyr/cmake/app/boilerplate.cmake
-- Application: /app/zmk/app
-- Adding ZMK config directory as board root: /app/config
-- ZMK Config directory: /app/config
-- Using keymap file: /app/config/adv360_left.keymap
-- Zephyr version: 3.0.0 (/app/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: adv360_left
-- Cache files will be written to: /root/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.13.2 (/opt/zephyr-sdk-0.13.2)
-- Found dtc: /opt/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.2 (/opt/zephyr-sdk-0.13.2)
-- Found BOARD.dts: /app/config/boards/arm/adv360/adv360_left.dts
-- Found devicetree overlay: /app/config/adv360_left.keymap
-- Generated zephyr.dts: /app/build/left/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /app/build/left/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /app/build/left/zephyr/include/generated/device_extern.h
[... 50 lines cut ...]
/app/zmk/app/include/zmk/split/bluetooth/central.h:11:35: warning: 'struct backlight_state' declared inside parameter list will not be visible outside of this definition or declaration
   11 | int zmk_split_bt_update_bl(struct backlight_state *periph);
      |                                   ^~~~~~~~~~~~~~~
[64/344] Building C object CMakeFiles/app.dir/src/backlight.c.obj
In file included from /app/zmk/app/src/backlight.c:26:
/app/zmk/app/include/zmk/split/bluetooth/central.h:10:36: warning: 'struct zmk_periph_led' declared inside parameter list will not be visible outside of this definition or declaration
   10 | int zmk_split_bt_update_led(struct zmk_periph_led *periph);
      |                                    ^~~~~~~~~~~~~~
[65/344] Building C object CMakeFiles/app.dir/src/rgb_underglow.c.obj
In file included from /app/zmk/app/src/rgb_underglow.c:35:
/app/zmk/app/include/zmk/split/bluetooth/central.h:11:35: warning: 'struct backlight_state' declared inside parameter list will not be visible outside of this definition or declaration
   11 | int zmk_split_bt_update_bl(struct backlight_state *periph);
      |                                   ^~~~~~~~~~~~~~~
[109/344] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/storage/flash_map/flash_map.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/subsys/storage/flash_map/flash_map.c.obj
ccache /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/app/zephyr/kernel/include -I/app/zephyr/arch/arm/include -I/app/zephyr/include -I/app/build/left/zephyr/include/generated -I/app/zephyr/soc/arm/nordic_nrf/nrf52 -I/app/zephyr/soc/arm/nordic_nrf/common/. -I/app/zephyr/subsys/bluetooth -I/app/zephyr/subsys/usb -I/app/zephyr/subsys/settings/include -I/app/modules/hal/cmsis/CMSIS/Core/Include -I/app/modules/hal/nordic/nrfx -I/app/modules/hal/nordic/nrfx/drivers/include -I/app/modules/hal/nordic/nrfx/mdk -I/app/zephyr/modules/hal_nordic/nrfx/. -I/app/modules/crypto/tinycrypt/lib/include -I/app/zmk/app/drivers/sensor/battery/. -isystem /app/zephyr/lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -Os -imacros /app/build/left/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /app/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/app/zmk/a [... cut ...] /flash_map.c
arm-zephyr-eabi-gcc: internal compiler error: Segmentation fault signal terminated program cc1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
[114/344] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/usb/usb_device.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /app/build/left
make: *** [firmware/$(TIMESTAMP)-left.uf2] Error 1
```

gggdomi, 2022-10-25:
I managed to build using ZMK 3.0 image (chosen because it has arm64 arch [here](https://hub.docker.com/layers/zmkfirmware/zmk-build-arm/3.0-branch/images/sha256-29eeaa1e7033222e808c4a00ac12d3f0e99ba60ff5dbbd27f59fe12dd49adefb?context=explore))

```diff
Dockerfile
+ FROM zmkfirmware/zmk-build-arm:3.0-branch
- FROM zmkfirmware/zmk-build-arm:stable
```

I'm not sure if I can flash the keyboard with this v3.0 firmware? Is it supposed to be supported yet?

ReFil, 2022-10-25:
This is an issue I will raise with ZMK team, thanks for bringing it to my attention. You should be able to use that image but the stable needs updating

ReFil, 2022-10-27:
Can confirm upstream issue now fixed

closed 2022-10-27 by ReFil