KinesisCorporation/Adv360-Pro-ZMK #40: LEDs are always on and display battery status

afilipchik, 2022-10-27:
Happened to 1 keyboard and was able to reproduce with the second one. 
 
First keyboard got in a mode when all 6 LEDs are always on and display battery state by. Now I think office cleaning stuff accidentally pressed a magic combination.  I later was able to repro by pressing Mod + HotKey 2 and 3 on my second keyboard, and it is impossible to disable. Mod+Space to turn LEDs off works, but they only display battery state.

Playing with different firmware versions didn't help

KinesisCorporation, 2022-10-27:
Sorry to hear this. We have seen a couple instances of this but have not been able to repro it here using any combination of simulatenous Mod commands. We are actively investigating the issue.

Mod + HK2 and HK4 will cause the battery status to stick, but power cycling fixes it. We have removed the Mod + HK2 command from new builds to eliminate the risk of this particular issue.

ReFil, 2022-10-27:
Hiya, a solution to fix this is map &rgb_ug RGB_EFS 4 to an unused key on the mod layer then press it. This will reset the lighting to the proper effect. Once done do not reset or power off the board for at least 2 minutes to ensure the flash timeout occurs and the new settings are saved to memory. This should only need to be done once

KinesisCorporation, 2022-10-27:
So the command needs to be added to both modules separately. We've created some repair files and step-by-step instructions and posted them here:

https://kinesis-ergo.com/support/kb360pro/#troubleshooting. 

Please let us know if this works.

afilipchik, 2022-10-28:
Silly question: @ReFil what is &rgb_ug RGB_EFS 4? I Only see RGB_EFF and RGB_EFR via visual editor.
Should I just manually add RGB_EFS, and then what will be "4"?

afilipchik, 2022-10-28:
@KinesisCorporation repair firmware works, thank you! also, once i flushed it, I only needed to reset 1 module, and second also fixed itself right away

KinesisCorporation, 2022-11-07:
See fix above

closed 2022-11-07 by KinesisCorporation