KinesisCorporation/Adv360-Pro-ZMK #41: FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /app/build/left

zemackdaddy, 2022-10-27:
internal compiler error: Segmentation fault signal terminated program cc1
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /app/build/left

MacOS Monterray
M1 Max 2016

Can someone point me in the right direction?
Or is my Mac not supported?

I cloned from https://github.com/KinesisCorporation/Adv360-Pro-ZMK v2.0.

Then ran `make all`.

`[68/344] Building C object CMakeFiles/app.dir/src/rgb_underglow.c.obj
In file included from /app/zmk/app/src/rgb_underglow.c:35:
/app/zmk/app/include/zmk/split/bluetooth/central.h:11:35: warning: 'struct backlight_state' declared inside parameter list will not be visible outside of this definition or declaration
   11 | int zmk_split_bt_update_bl(struct backlight_state *periph);
      |                                   ^~~~~~~~~~~~~~~

[69/344] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj

FAILED: zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj 
ccache /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/app/zephyr/kernel/include -I/app/zephyr/arch/arm/include -I/app/zephyr/include -I/app/build/left/zephyr/include/generated -I/app/zephyr/soc/arm/nordic_nrf/nrf52 -I/app/zephyr/soc/arm/nordic_nrf/common/. -I/app/zephyr/subsys/bluetooth -I/app/zephyr/subsys/usb -I/app/zephyr/subsys/settings/include -I/app/modules/hal/cmsis/CMSIS/Core/Include -I/app/modules/hal/nordic/nrfx -I/app/modules/hal/nordic/nrfx/drivers/include -I/app/modules/hal/nordic/nrfx/mdk -I/app/zephyr/modules/hal_nordic/nrfx/. -I/app/modules/crypto/tinycrypt/lib/include -I/app/zmk/app/drivers/sensor/battery/. -isystem /app/zephyr/lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include -isystem /opt/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/include-fixed -Os -imacros /app/build/left/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /app/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/app/zmk/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/app/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/app=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -Wfatal-errors -MD -MT zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj -MF zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/lib/os/crc7_sw.c.obj -c /app/zephyr/lib/os/crc7_sw.c
arm-zephyr-eabi-gcc: 

internal compiler error: Segmentation fault signal terminated program cc1

Please submit a full bug report,
with preprocessed source if appropriate.

See <https://gcc.gnu.org/bugs/> for instructions.

[75/344] Building C object CMakeFiles/app.dir/src/split/bluetooth/central.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /app/build/left
make: *** [firmware/$(TIMESTAMP)-left.uf2] Error 1
`

ReFil, 2022-10-30:
M1 Macs encountered issues with the zmk docker image. This has now been fixed by the ZMK team https://github.com/zmkfirmware/zmk-docker/issues/126

Please test again after fetching latest docker image and let me know if the problem is fixed and i can close out this issue

zemackdaddy, 2022-10-30:
Works fine now.  Thank you so much.

closed 2022-10-30 by zemackdaddy