KinesisCorporation/Adv360-Pro-ZMK #43: Flashing fails with "unexpected error" on macOS 13

gohanlon, 2022-11-02:
I get the following error when attempting to copy either my custom firmware or the `settings-reset.uf2` file (from the V2.0 branch) to my keyboard's mounted volume:

> The operation can't be completed because an unexpected error occurred (error code 100093).

The const name of that code is `kPOSIXErrorENOATTR = 100093`. I've tried rebooting my computer to no avail.

This is the first time I've tried flashing my Adv360 since updating the host OS to macOS 13.0 (Ventura), and I've successfully flashed my keyboard many times before updating to macOS 13. Considering that and the specific error code, it seems likely that something changed in the filesystem protocol across major macOS versions. Perhaps others can report whether they've successfully flashed while running macOS 13?

My keyboard is still running firmware built on the V1.0 branch. (This came up while rebasing my customizations from the V2.0 branch.)

sevenmac, 2022-11-02:
I had the same issue on my Ventura macOS. The problem is self flashing volumes and cp command combination.  Used the following commands to flash keyboard. It still throws error yet the flashing takes effect. cd into the firmware directory on terminal and run

```
rsync left.uf2 /Volumes/ADV360PRO/
rsync right.uf2 /Volumes/ADV360PRO/
```

gohanlon, 2022-11-02:
Thank you, @sevenmac and @ReFil. I successfully flashed using `cp` rather than the macOS Finder.

@ReFil, I'll leave it to you whether to close this issue or leave it open to track the upstream issue.

davidlashlee, 2022-11-04:
came to say i also ran into this. Coudlnt transfer file via finder. was successful in flashing via cp even though i got errors. ill try rsync next time.

davidlashlee, 2022-11-05:
some interesting info, when self building the firmware i could transfer without issues, but the github actions firmware still shows errors.

ReFil, 2022-11-08:
This issue is also affecting all raspberry pi picos so i'd expect an OS X update to fix it soon

ReFil, 2022-11-17:
Just clarifying with the latest updates on this issue, it has been written up the raspberry pi foundation and they're communicating with apple to sort this issue, adafruit have also written it up . It causes problems for a lot of embedded devices that present a "fake" volume to the computer. In the mean time `rsync` seems to be a good workaround, `cp` seems to be a bit flakier although should work fine with uf2 files. 

https://www.raspberrypi.com/news/the-ventura-problem/
https://blog.adafruit.com/2022/10/31/uploading-uf2-files-with-macos-13-0-ventura-apple-microbit_edu-raspberry_pi-circuitpython/

ReFil, 2022-12-07:
According to [this comment](https://github.com/ARMmbed/DAPLink/issues/982#issuecomment-1339852504) the issue seems to be better on the 13.1 beta4, is anyone able to test this?

ReFil, 2022-12-14:
Fixed on ventura 13.1. Can someone please test this so i can close out this issue as completed

KinesisCorporation, 2022-12-15:
Working on our end with Ventura 13.1. Nice job Apple!

closed 2022-12-15 by KinesisCorporation