KinesisCorporation/Adv360-Pro-ZMK #45: Is it possible to customize layer colors?

sharwell, 2022-11-05:
By default, the 'kp' layer lights up the layer LED as white, 'fn' lights it as blue, and 'mod' lights it as green. Is it possible to customize the color of this LED and/or assign colors to other layers?

ReFil, 2022-11-07:
The layer colours were not designed to be changed, although it can be done by forking my adv360_z3 branch, redirecting the user config repo to point to your fork and changing the individual r g and b values in app/src/rgb_underglow.c. The led indicators function mentioned in those docs is not referring to the layer indication, it's a different subsystem. I've implemented user configurable per layer lighting for a different project before (https://github.com/ReFil/zmk/commit/2f7c0420b386637b630693398d28aa4beed6c898)  but don't have the time to port it to the adv360 branch. If anyone else is willing to do so I'd be more than happy to pr it in, it shouldn't be too hard and won't cause issues

ReFil, 2022-11-15:
Closing as changing this is not in the roadmap

closed 2022-11-15 by ReFil