KinesisCorporation/Adv360-Pro-ZMK #46: can't flash onto the right side

agzam, 2022-11-05:
I can't flash the latest v2.0 754f13303d86c91126c2d64c536a773b77615285 onto the right side (even though the left side seems to work fine). After copying the file, the unit keeps flashing red. And then won't connect when turned on.

dylan-jh, 2022-11-06:
You should be able to resync the modules by powercycling, left then right. If not, just connect the right side to the PC and use a paperclip to open the bootloader drive and transfer right.uf2.

KinesisCorporation, 2022-11-06:
You should be able to resync the modules by powercycling, left then right. If not, just connect the right side to the PC and use a paperclip to open the bootloader drive and transfer right.uf2.

agzam, 2022-11-06:
I'm not sure what was causing it, for a moment I thought I bricked the unit. It took me a while to figure it out. First I thought there's something wrong with my settings (first created on v1.0) rolled on top of the latest v2.0. I tried different combinations, but none of them would work.

Finally, I flashed `settings-reset.uf2`, and realized that flashing it mounts the drive, copying the firmware build that way have worked.

closed 2022-11-06 by agzam