KinesisCorporation/Adv360-Pro-ZMK #47: Custom behaviors in keymap editor

tyronep, 2022-11-06:
In a future release, might the team consider making custom behaviors, such as the `&hm` behavior supplied in the default configuration, selectable in the keymap editor? It's a small job to manually edit the generated config file after each commit, but one I'd like to avoid, especially while iterating on layouts. Thanks!

KinesisCorporation, 2022-12-12:
Can you link to the documentation on &hm. We're not finding it.

tyronep, 2022-12-12:
@KinesisCorporation `&hm` specifically is predefined for users in [`adv360.keymap`](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V2.0/config/adv360.keymap#L11). 

Supporting mapping of other user-defined modifiers would also be ideal, but even implementing `&hm` (or any other Kinesis-defined modifier) in the GUI would be immediately useful.

KinesisCorporation, 2022-12-13:
Gotcha. So you are looking for  menu in the GUI that would allow you to change the Hold-Tap settings?

- tapping-term-ms = <200>;
- quick_tap_ms = <175>;
- flavor = "tap-preferred";

ReFil, 2023-01-12:
Closing as unplanned, the keymap editor is outside the scope of this repo

closed 2023-01-12 by ReFil