KinesisCorporation/Adv360-Pro-ZMK #48: FATAL ERROR: command exited with status 1

P05TPWND, 2022-11-06:
I'm having a different issue than others have alluded to I think (previously, it seemed due to Docker):

```Run west build -s zmk/app -d build/left -b adv360_left -- -DZMK_CONFIG="${GITHUB_WORKSPACE}/config"
[2](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:2)
  west build -s zmk/app -d build/left -b adv360_left -- -DZMK_CONFIG="${GITHUB_WORKSPACE}/config"
[3](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:3)
  shell: sh -e {0}
[4](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:5)
-- west build: generating a build system
[5](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:6)
Including boilerplate (Zephyr base): /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr/cmake/app/boilerplate.cmake
[6](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:7)
-- Application: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zmk/app
[7](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:8)
-- Adding ZMK config directory as board root: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/config
[8](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:9)
-- ZMK Config directory: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/config
[9](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:10)
-- Using keymap file: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/config/adv360_left.keymap
[10](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:11)
-- Zephyr version: 3.0.0 (/__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr)
[11](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:12)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
[12](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:13)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
[13](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:14)
-- Board: adv360_left
[14](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:15)
-- Cache files will be written to: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr/.cache
[15](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:16)
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
[16](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:17)
-- Found host-tools: zephyr 0.13.2 (/opt/zephyr-sdk-0.13.2)
[17](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:18)
-- Found dtc: /opt/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
[18](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:19)
-- Found toolchain: zephyr 0.13.2 (/opt/zephyr-sdk-0.13.2)
[19](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:20)
-- Found BOARD.dts: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/config/boards/arm/adv360/adv360_left.dts
[20](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:21)
-- Found devicetree overlay: /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/config/adv360_left.keymap
[21](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:22)
devicetree error: /keymap/layer_keypad: undefined node label 'macro_copy'
[22](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:23)
CMake Error at /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr/cmake/dts.cmake:205 (message):
[23](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:24)
  gen_defines.py failed with return code: 1
[24](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:25)
Call Stack (most recent call first):
[25](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:26)
  /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr/cmake/app/boilerplate.cmake:542 (include)
[26](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:27)
  /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
[27](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:28)
  /__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
[28](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:29)
  CMakeLists.txt:15 (find_package)
[29](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:30)

[30](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:31)

[31](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:32)
-- Configuring incomplete, errors occurred!
[32](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:33)
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/build/left -S/__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/zmk/app -GNinja -DBOARD=adv360_left -DZMK_CONFIG=/__w/Adv360-Pro-ZMK-NP/Adv360-Pro-ZMK-NP/config
[33](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/3405595447/jobs/5663688359#step:8:34)
Error: Process completed with exit code 1.```

I think I made a config that's forcing the build to fail, but I'm unsure how to revert it. Nearly all subsequent builds are failing. This is really my first go-round with ZMK, so I've been using the GUI. Can anyone help? Happy to provide any details that may assist. I'm not sure exactly what's causing the error to be thrown with exit code

jkowall, 2022-11-06:
Why not use Gh actions? Less stuff to deal with and automates the build upon pushing changes to a branch.

P05TPWND, 2022-11-06:
How do I revert a GH action to a known working state?

P05TPWND, 2022-11-06:
The problem I'm facing is now every subsequent change I try to make here on out forces the GH build to fail. I think worse case, I can delete my fork and create a whole new one and basically start over, but I'm trying to understand what the problem even is so I don't have to do that again

jkowall, 2022-11-06:
If the compilation doesn't work on GH actions, it's not going to work in Docker. I would suggest deleting your fork and re-forking it. I would then create a branch for your edits (each time). This makes it easier to re-sync the upstream to the main fork and keep your changes separate. Just make your edits in smaller chunks and if the compilation fails, then you've made an invalid change.

P05TPWND, 2022-11-06:
I was overthinking this and resolved it. IDK what I changed that made it fail in my keymap, but I found the last known good edits to these files and recommitted them. Now it's fixed. I like the branch idea and may use that for experimentation as I learn more about ZMK.

closed 2022-11-06 by P05TPWND

jkowall, 2022-11-06:
> it's just got, you don't need to make a branch for each change, or "refork" anything... Commit each change, use git reset. Add this repo as a remote and diff the V2.0 branch with your branch, etc...

I prefer to make edits in a branch, you can resync the fork as you wish. If you do git reset you'll lose your changes each time. Managing diffs versus PRs is not ideal IMO.

P05TPWND, 2022-11-06:
> I prefer to make edits in a branch, you can resync the fork as you wish. If you do git reset you'll lose your changes each time. Managing diffs versus PRs is not ideal IMO.

This is a great idea honestly. For now, I'm going to use a permanent "fall back" branch I can merge to revert more quickly. This is mostly because I'm using the GUI for now until I've fully understood the documentation to make my own in-file changes without GUI.