KinesisCorporation/Adv360-Pro-ZMK #49: Unable to use keyboard via Bluetooth while charging left unit with wall charger

wojo, 2022-11-08:
Something about charging the left unit via a standard wall charger (not a computer) causes the Bluetooth connection to go offline and assumes that it should do communication over USB I think. This results in a keyboard that is not usable when plugged into a standard (dumb) wall charger.

Is this intended? I'm on the latest V2.0 version as of Nov 8 2022.

KinesisCorporation, 2022-11-08:
It's not intended to be charged except by a PC.

wojo, 2022-11-08:
> It's not intended to be charged except by a PC.

Ah, interesting. Is this a ZMK limitation or part of the hardware?

Either way, probably can close this then.

arogulin, 2022-11-10:
@wojo this appears to be ZMK limitation and default configuration to prefer USB output when both USB and BLE are connected because `ZMK is not always able to detect if the other end of a USB connection accepts keyboard input or not`: https://zmk.dev/docs/behaviors/outputs
The good news is that you can override that behavior via firmware if needed.

wojo, 2022-11-10:
@arogulin perfect, thanks for pointing that out. This is why going with the ZMK pro keyboard is nice, I can change the behavior!

closed 2022-11-15 by ReFil