KinesisCorporation/Adv360-Pro-ZMK #54: I cannot type grave any more; only getting §

zemackdaddy, 2022-11-14:
My macOS keyboard layout is US.

Everything was fine until I flashed the keyboard today.

Happens locally as well as with the GitHub actions.

[My fork](https://github.com/zemackdaddy/Adv360-Pro-ZMK/tree/V2.0)

Perhaps the issue is in the zmk repo?  I wouldn't know where to even look.

TravisHi, 2022-11-14:
The key before your grave is missing an `&`? I've never used ZMK before, but it looks like a typo? `MO_TOG(2)` should it be something like `&MO_TOG(2)`?

https://github.com/zemackdaddy/Adv360-Pro-ZMK/blob/V2.0/config/adv360.keymap#L78

zemackdaddy, 2022-11-14:
For the MO_TOG, you need to look at the definitions at the top of the file.

I don't think that has anything to do with &kp GRAVE outputting on international char which is usually a keyboard layout issue.

zemackdaddy, 2022-11-15:
This only happens when the keyboard is plugged in.  When it is not, the ` outputs properly.

ReFil, 2022-11-15:
Wait, so the key does different things depending on whether it's connected to the same device over bluetooth or USB?

zemackdaddy, 2022-11-15:
Well, I swear that is what was happening last night but now, it is constantly outputting §.

ReFil, 2022-11-15:
This seems like it might be a software thing with macs, i know they like to kinda screw with key mappings, I've been doing keymap testing on macs recently and have encountered numerous strange behaviours around less used keys

zemackdaddy, 2022-11-15:
I wonder if macOS isn't able to detect the keyboard type properly?

In system preferences > keyboard, "change keyboard type" will not detect it and I have to choose "ANSI North America", which doesn't fix the issue.

ReFil, 2022-11-15:
Maybe try forcing a few different layouts in system preferences and see what happens?

zemackdaddy, 2022-11-15:
I already tried some non-english layout.  Didn't work.

I've even tried Karabiner-Elements to remap it.
 
This wasn't a problem until Monday.

zemackdaddy, 2022-11-15:
I should state that the Mac keyboard viewer shows the proper tilde.  That is why I don't think it is the OS.

zemackdaddy, 2022-11-15:
Restarting took care of it. Sorry to waste your time.

ReFil, 2022-11-15:
No worries, glad to hear it's sorted now

zemackdaddy, 2022-11-15:
Spoke too soon.  Now it is typing the ` vs § when I its plugged in again.

ReFil, 2022-11-15:
This definitely seems like some sort of software issue on the mac, is this behaviour replicable on a computer with a different OS?

zemackdaddy, 2022-11-15:
I disconnected the 360 pro from my Mac and paired it with a Dell laptop.

Exact same issue.

Makes me think it is the firmware on the keyboard.

ReFil, 2022-11-15:
Ok that's really strange. that's not an issue i can replicate. &kp GRAVE always gives me ` and ~ with shift held, tried over usb and bluetooth on my dell latitude running ubuntu

ReFil, 2022-11-15:
If you flash back to default firmware what happens?

KinesisCorporation, 2022-11-15:
Can you try reverting back to default firmware and re-test?

https://kinesis-ergo.com/support/kb360pro/#firmware-updates

zemackdaddy, 2022-11-15:
I flashed the settings_reset to both sides and then the left/right uf2s.

Same problem.

I know the firmware reset took because I had to re-pair to my Mac and my Dvorak layout was gone (obviously).

Sorry.  I know it isn't what we want to hear but the default firmware also has the problem.

I'm the only one?

zemackdaddy, 2022-11-15:
I have to admit the settings reset flash confuses me.  After flashing the left, it just sits there and blinks so I disconnect the cable, flash the right, disconnect the cable, then connect the left and flash the default firmware.  Then the right.

If I did that right, the problem doesn't go away.

ReFil, 2022-11-15:
Are you setting the keyboard to Dvorak in the OS? I haven't seen this issue mentioned by anyone else

zemackdaddy, 2022-11-15:
I am not. My keymap in my repo is. When flashing your default settings and firmware that mapping is gone. I am not messing around with the OS settings for keyboards in any way.

This email was sent to you by MetaStock. Any views expressed in this message are those of the individual sender, except where the sender specifically states them to be the views of MetaStock.

ReFil, 2022-11-15:
I see, honestly I don't know what it could be

KinesisCorporation, 2022-11-15:
We'll do some experimentation on our end and get back to you.

zemackdaddy, 2022-11-15:
Good news:  I went back and reapplied the settings reset and
default firmware, paired with windows, and get the ` when not plugged in.
This time, I realized that in earlier tests when plugging in the cable
while paired to Dell, that the cable was connected to the Mac.  So, sloppy.

*When paired to the dell or connected to it correctly via cable, the ` is
correctly output.*

It appears that even with the default firmware and settings reset that *just
being plugged into the M1 2021 16" Monterey Mac, causes the § to be output*.

So, OS issue.  But why only when plugged in?
I've only been using the keyboard not attached via cable for about a week.
Previously, always.
And I use the ` and ~ constantly and noticed the second it stopped working
yesterday.

KinesisCorporation, 2022-11-15:
Hmm... Still trying to wrap our heads around this new development.

So on your current build linked above
- Connected over USB to Mac Grave outputs §
- Connected over BT to Mac Grave outputs Grave
- Connected over either BT or USB to a PC Grave outputs Grave

Do we have that right?

zemackdaddy, 2022-11-15:
Correct.

Apologies for the sloppy testing earlier.

It is a Mac issue.  I think it is related to the OS not being able to
detect the keyboard type but what do I know?

Sure wary of updating to Ventura but maybe that would fix it :)

zemackdaddy, 2022-11-16:
So, I found people saying I should clear the NVRAM/SMC.

According to this, you turn an M1 off and let it sit for 30 sec.  I did and the problem is fixed for now.

Doesn't mean it wasn't some odd misunderstanding between macOS and the keyboard, but it is something I can do if it is infrequent enough.

closed 2022-11-16 by zemackdaddy

ReFil, 2022-11-16:
Thats so strange that clearing NVram fixed it, but if it's fixed thats good

KinesisCorporation, 2022-11-17:
Just tested your firmware on a Ventura Mac here and Grave outputs Grave over USB... Not sure what's happening but it looks like its specific to your Mac or settings. Keep us posted.

zemackdaddy, 2022-11-17:
Nothing to post.  I've closed the issue.  Mac SMC or something like that
was the issue.  Not sure how, but it was.

equiman, 2024-02-08:
I've solved the problem changing the keyboard type from ISO to ANSI. `Settings -> Keyboard -> [Change Keyboard Type]` change to ANSI, and using the `Keyboard viewer` you can see if was correctly setted.

ANSI:
[image]

ISO:
[image]

> Some times need to be setted twice to apply the changes.

luiscmartinez, 2024-07-09:
Hello, I had the same problem aka when pressing the grave key, I was only getting § symbol.
Here are the steps I took to fix it. First, restart my mac. Second, go to settings > keyboard > change keyboard type... 
This starts the keyboard setup assistant. I just continued thru it and at the end grave key went back to normal inputting the tidal

max-si-m, 2024-08-13:
@equiman I have been struggling with this issue for a year already, but I didn't know how to fix it. Man, thank you, changing the keyboard type helped. 🤗🤗🤗