KinesisCorporation/Adv360-Pro-ZMK #60: Can't connect to Mac via bluetooth

humblehacker, 2022-11-17:
I flashed my keyboard with firmware built from the current head (2562f62), and now I can no longer connect my keyboard via bluetooth. It correctly shows up as "Adv360 Pro" under nearby devices, but when I press "Connect", I get a spinner, and briefly the button changes to "Disconnect" then immediately switches back to "Connect".

In my Console logs I get a constant stream of the following:
```
2022-11-17 15:34:47.670742 -0800	info	bluetoothd	com.apple.bluetooth	CBStackBLEScanner	Device found changed: CBDevice D06048F3-C746-C12A-EC6F-D4C778F201ED, BDA E0:07:2A:6F:CE:0E, Nm 'Adv360 Pro', DsFl 0x2000000000000 < BLEHID >, RSSI -57, Aprc Keyboard, CF 0x200000000 < RSSI >
2022-11-17 15:34:47.670960 -0800	debug	bluetoothd	com.apple.bluetooth	Server.LE.Scan	Found device "D06048F3-C746-C12A-EC6F-D4C778F201ED with data:"Adv360 Pro", RSSI: 0 dB (non-saturated), Service UUIDs: 0x1812 0x180F, connectable, sourceCore: MainCore, IsELNAOn: 0, IsPassup: 0, IsFromSCCompensation0,  Setting name to Adv360 Pro
```

I am on the Ventura 13.1 beta, but the keyboard has been working fine with that so far.  I'm going to attempt to flash one of my previous firmware builds.

humblehacker, 2022-11-17:
No luck. I reflashed using my previously working firmware built last month, but it's exhibiting the same problem.

KinesisCorporation, 2022-11-18:
Hi there.

Best bet is to re-pair from scratch.
1) Forget/Disconnect the keyboard from your PC
2) Use the Mod + Windows key command to disconnect the PC from your keyboard

Profile LED should resume flashing quickly and you should be able to make a fresh connection.

If that doesn't work, try pairing one of the other 4 channels with a different device like your smart phone and report back.

humblehacker, 2022-11-18:
Back in business. I turned off both halves, reflashed each with `settings-reset.uf2`, reflashed again with my newest firmware build, turned on the left, then turned on the right. After that I was able to pair the keyboard in Bluetooth settings.

Not sure what I did to get into this state, but I'm glad the reset fixed it.

humblehacker, 2022-11-18:
@KinesisCorporation Thanks for the quick response! I did all of that except the Mod+Windows key. I missed the "Bluetooth clear" section of the manual. That probably would have done the trick easier than the reset I just performed. But problem solved either way, so thanks again!

closed 2022-11-18 by humblehacker