KinesisCorporation/Adv360-Pro-ZMK #64: LED indicators (caps|num|scroll)lock not working on BLE

laserscout, 2022-11-22:
I'm looking into this currently, just wanted to make sure it's not just me on my mac.

The CAPSLOCK `LEFT 0`, NUMLOCK `RIGHT 0` and SCROLLLOCK `RIGHT 1` indicators are only operational when USB is connected; right?

laserscout, 2022-11-22:
This code added in ReFil's zmk fork seems though like it's supposed to do just that?

https://github.com/ReFil/zmk/commit/aa62fd391d679c46e9e6f35b2225719d129a6ff9?diff=unified#diff-b3e2c5329969f421f8e1dc4e06b59b38e157d9555b2ed0557dc55ce6c9f35097R95-R115

Maybe @bortoz as the author can clarify?

bortoz, 2022-11-23:
It's a bug that has already been fixed but the code in this repo hasn't been updated.

The fix is [this](https://github.com/zmkfirmware/zmk/compare/37a79ac0ca540bb27f403f6be8e39d2c1d72c772..9eae1187a1b8cdf3e5cfa5e6c847a8aa6ef1b882).

Comment in the original PR: https://github.com/zmkfirmware/zmk/pull/999#issuecomment-1287828950.

laserscout, 2022-11-23:
I had already tried that in my [led-indicators branch](https://github.com/laserscout/zmk/commits/led-indicators) which was based on your latest with Refil's `adv360-z3` necessary commits rebased on top. I tried to make it work yesterday, but it didn't work.

So, thank you for also commenting the PR link which indicated I should **unpair** and repair, hence fixing the issue!

Hopefully the @ReFil "official" zmk fork for the Adv360 will get your latest changes merged, and we can all enjoy some Christmas like lights on our keyboards 😝 

Buonasera

ReFil, 2022-11-23:
Will patch in the latest led indicators asap

ReFil, 2022-11-23:
This is a core zmk thing, I've heard it mentioned before. I think there's an open PR to add a BIOS mode

ReFil, 2022-11-25:
Just pushed a quick hotfix to fix the bug here: https://github.com/ReFil/zmk/commit/20782cdb48fe5be1c9fb7e17068d1b3c6bd7dd87

to use it either rerun the build job in github actions or run make clean and make if building locally

closed 2022-11-25 by ReFil