KinesisCorporation/Adv360-Pro-ZMK #65: Swapping bluetooth profiles after turning off keyboard to connect to a new device fails

TravisHi, 2022-11-22:
1. Added to initial device using KA360 bluetooth profile 1,
2. Swap to second device,
3. Added second device using KA360 bluetooth profile 2,
4. Swap back to initial device,
5. Right half no longer connects to left half, on either the initial or second device, in USB or bluetooth modes (Can't change BT profile because the mod key is on the right half).
6. Had to re-flash both halves to regain right to left connection.

Hey guys I seem to get locked into a no-mans land if I try to toggle between bluetooth profiles on separate devices.

KinesisCorporation, 2022-11-22:
Hi Travis,

Sorry to hear you are having trouble. That doesn't _sound_ like normal behavior. You should be able to hold Mod down and cycle between any of the 5 profiles without any issue regardless of pairing setup.

We just installed your latest build and tested it here and it works fine using the default Mod location. No problems switching between profiles with different devices.

If you want to upload a quick video we'd love to see it happening first hand. Then we can go about issuing an RMA for Repair via our tech portal.

https://kinesis-ergo.com/support/contact-a-technician/

KinesisCorporation, 2022-11-23:
Glad to hear it. The right can sometimes lose track of the left so it may have been totally unrelated to switching profiles.

Usually a simultaneous power-cycle fixes any left-right sync issues. Best to try that first before reflashing if it happens again. We'll close issue but feel free to re-open or contact our Tech Support directly if it crops up again.

closed 2022-11-23 by KinesisCorporation

FlippingBinary, 2022-12-22:
I'm not sure if this is related, but I might have stumbled on a repeatable series of steps that consistently disconnect the right module from the left in a way that is related to Bluetooth pairing and sounds similar to what is described in the title. 

1. Press `[Mod] 3` (to switch to an unpaired Bluetooth profile).
2. While the profile light is rapidly flashing, flip the power switch of each module off.
3. Count to 20 (or 10 if you're impatient).
4. Flip the power switch of each module on.
5. Observe the right module's three LEDs blinking red.
6. Reconnect the modules by using the reset button after you've exhausted all other options and grown tired of having a left handed keyboard.

Can someone else confirm if they are able to repeat this experience?