KinesisCorporation/Adv360-Pro-ZMK #69: macro doesn't work

shiqicao, 2022-12-01:
I added a macro and assign it to HK 3, but after build and update. Nothing happens if I press HK 3. Is there something I did wrong? Here is my change, https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/ff9a7fcf0f3f148cbd767d12a9ad6f0ca6974cc1

Thanks

dylan-jh, 2022-12-02:
It looks like you may have assigned the macro to the position of HK4 (the lowest key in the inner column of 3). Can you test that?

shiqicao, 2022-12-02:
doesn't work, I pressed "4", which is on the left side of "H"

KinesisCorporation, 2022-12-02:
Unfortunately we're not able to troubleshoot individual builds. Best approach is to start totally from scratch with a one-key macro to see if that works. Then make one change at a time to see where things go wrong.

shiqicao, 2022-12-02:
@KinesisCorporation thanks for replying, would you give me an example of how to configure a macro, for example, you can send me a git branch, where just a macro is added. Then I can try to test it on my side. Thanks!

KinesisCorporation, 2022-12-02:
There are some example macros in the default firmware. You should try assigning one of those to HK3 and then try editing that macro either through the configuration files or the GUI. Good luck!

We were able to get your password macro working last night so it should be a simple matter once you figure out what you are doing wrong.

shiqicao, 2022-12-02:
by the way, when a macro is assigned to HK3(the key left to Y), do I press only HK3, or I need to press some modifier key?

KinesisCorporation, 2022-12-02:
Just hotkey 3

shiqicao, 2022-12-02:
previously I only updated firmware for right one, this seems the problem. After updated firmware for both, macro works!

ReFil, 2022-12-02:
Keymap changes need both sides updating. Closing as solved

KinesisCorporation, 2022-12-02:
Yeah, you need to install the update on both left and right modules per all of the instructions. Glad you got it working.

closed 2022-12-02 by KinesisCorporation