KinesisCorporation/Adv360-Pro-ZMK #72: Num Lock status LED only works if the left side is plugged in to the PC

aidedinertial, 2022-12-05:
With both sides connected via bluetooth, the Num Lock status LED does not turn on.

It appears that the functionality of num lock is correct, just that the status LED doesn't work. I have my Advantage2 also connected, and I can see the status light change on that keyboard, but not on the Adv360.

ReFil, 2022-12-05:
Does the LED work over USB? A recent bugfix was pushed to improve the reliability of the LEDS over bluetooth, have you rebuilt your firmware lately?

aidedinertial, 2022-12-05:
Yes, the LED works over USB.

I put my keymap customization on the head of `V2.0` @ 0fb8e5824fee2fb11f263de745f5b1c0efbcd78a and built firmware locally with WSL2/docker.

aidedinertial, 2022-12-05:
@ReFil `make clean` had no effect on this behavior. Bug still exists.

Thanks for your fast feedback!

ReFil, 2022-12-05:
I see, did you flash both sides? It might help to unpair in windows and on the device and pair again

aidedinertial, 2022-12-05:
The keymap was updated with my changes, so I know that the FW load was successful.

FW Load procedure:
- Connect Left module via USB
    - Put Left into bootloader via Mod+1
    - Drag and drop left FW onto adv360 folder
 - Connect Right module via USB
     - Put Right module into bootloader via Mod+3
     - Drag and drop right FW onto adv360 folder
- Cleanup
    - Reset Right module via thumb cluster button
    - Reset Left module via thumb cluster button
    - Power the Right module Off
    - Power cycle Left module Off/On
    - Power the Right module On
 
I'll try the unpair/repair next.

aidedinertial, 2022-12-05:
If it's helpful, here's the build log:

file: build.sh
```bash
set +x

make clean
make
```

At terminal:
```bash
$chmod +x build.sh
$./build.sh 2>&1 | tee build.log
```

Build output:
[build.log](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/10155714/build.log)

aidedinertial, 2022-12-05:
To re-pair the current profile (Profile 2), I did the following:
- Mod+Win to clear the current profile
- Delete the `Adv360 Pro` from Windows Bluetooth devices
- Add new device > `Adv360 Pro`

No fix. Bug still exists.

- Reboot the computer

No fix. Bug still exists.

ReFil, 2022-12-05:
I see, this definitely seems like a duplicate of issue #64 and the fix that was implemented did help on some operating systems, I dont have a win10 machine to test with but i can confirm that the status LEDS work over Bluetooth on Linux

aidedinertial, 2022-12-05:
@ReFil If there are any patches you want me try, push a branch on your fork and I'll give it a go. Thanks!

aidedinertial, 2022-12-09:
@KinesisCorporation have you been able to reproduce this issue?

KinesisCorporation, 2022-12-12:
Can confirm Num Lock LED is not working on Windows 10 over Bluetooh

ReFil, 2023-11-06:
I recently retested this on windows 10 22h2 and can confirm it's still a present issue. I think it's a windows issue with any sort of output reports over BLE, as it causes issues with getting and setting feature reports too. It seems to be any host->device communication over HID-over-GATT

ReFil, 2023-11-06:
This is potentially related: https://www.esp32.com/viewtopic.php?t=19076 possibly GATT permissions

ReFil, 2024-01-07:
This seems to have been fixed with our newest beta version @aidedinertial if you would be able to test this firmware and provide feedback it would be much appreciated. Please note it might need unpairing and re-pairing to the host to account for the changed HID descriptor.

[360pro-beta-1-6-24.zip](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/13854890/360pro-beta-1-6-24.zip)

aidedinertial, 2024-01-08:
@ReFil The num lock LED is working properly on the `V3.0` branch at commit `1728a660 : (Prefer tr to ${char^^}, which does not work on older bash versions (#303))`.

The .zip file you provided also works.

```
Edition	Windows 11 Pro
Version	22H2
Installed on	‎9/‎15/‎2022
OS build	22621.2861
Experience	Windows Feature Experience Pack 1000.22681.1000.0
```

ReFil, 2024-01-08:
Thank you for confirming it is working now, sorry it took so long to sort. I'm closing this issue out as solved now :)

closed 2024-01-08 by ReFil