KinesisCorporation/Adv360-Pro-ZMK #81: Can't connect 2 Kinesis 360 Pro to Mac Book Pro (Monterey)

afilipchik, 2022-12-20:
TLDR: Look like only 1 Kinesis360 pro can be in the list of bluetooth devices at any given moment of time.

For some reason I'm having issues maintaining 2 Bluetooth connections for Kinesis 360 Pro (one at work and one at home). 
If one kinesis is in the list of devices, the second never connects over bluetooth. What is weird, that the first one works. 

For example: if I first connect one at work, it will work fine and reconnect without issues. If I then try to connect one at home, the device looks connected in the Bluetooth manager, but LED on keyboard is blinking and no Input is possible unless a wire is connected.

Refresh of device list usually results in displaying an error: "Preferences Error. There was an error in Bluetooth preferences.", and leads to kinesis disappearing from the list. 

If I remove the work keyboard from the list, home one connects fine but not I need to do the same at work.

KinesisCorporation, 2022-12-20:
This is with a laptop that is moving between home and work, but connecting to a different Pro keyboard at each location?

afilipchik, 2022-12-22:
Yes, exactly. 

I tried it yesterday again, and here are steps:
1. Remove all the keyboards from Bluetooth devices
2. Connect home keyboard -> works fine
3. Went to the office, connected work keyboard -> works fine
4. Went back to home, tried to connect -> didn't work till I removed both keyboards and reconnected.
5. I named them differently, it didn't help
6. Upgraded to the latest Ventura OS -> didn't help

afilipchik, 2022-12-22:
thank you! I wonder whether it is some shared identifier.

ReFil, 2022-12-22:
The identifier that matters is the MAC address and that should be unique per device

dylan-jh, 2023-01-31:
@afilipchik 

Any update?

KinesisCorporation, 2023-01-31:
@afilipchik 

Any update?

afilipchik, 2023-02-01:
@KinesisCorporation still not working. Is there any suggested fix or some debugging procedure I can do?

ReFil, 2023-02-01:
The baseline code was recently updated to the latest ZMK main, would you be able to rerun your build jobs and update the firmware?

ReFil, 2023-02-01:
Something else you could try is changing the keyboard name property so the boardds have different names. Update and rerun build actions, then flash one of the two boards, then change this line https://github.com/afilipchik/Adv360-Pro-ZMK/blob/bafe24f5be9d2ddd03a64fd7963c8b51c833a099/config/boards/arm/adv360/Kconfig.defconfig#L9 to change the bluetooth advertised name of the board (no more than 16 chars permitted) and flash the other board

KinesisCorporation, 2023-02-28:
We tested this yesterday and had no problems. Connected 2 different keyboards to 3 different devices (Macbook Pro M1, 2018 Macbook Pro, and Windows 11 Desktop).

Both keyboards were able to switch between all 3 devices and send keystrokes at the same time.

afilipchik, 2023-03-20:
was traveling, but will try the suggested things this week. 

Thank you!

robertkluin, 2023-04-19:
Has anyone found a fix for this?

I hit this same issue using two Kinesis 360 Pros with one Mac (14" Apple M1 Pro, OS X 13.3). I've needed to only use the wired connection. I've tried renaming them, resetting bluetooth (and removing the pairings), and using different profiles on the keyboards—no luck. I can pair my office keyboard, then my home keyboard will not reconnect. If it is helpful for the kinesis team, I encounter this after using one keyboard for a few hours then moving to a different location (the machine sleeps). If I don't attempt to connect to the other keyboard it will work fine.

ReFil, 2023-06-02:
We've developed a new major update for the Adv360 Pro with significant improvements to the bluetooth code especially with regards to mac compatibility and split connection reliability.
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/tree/V3.0-beta
It would be very helpful if people encountering issues with macs and split connectivity could test this branch. Our initial testing on our devices has so far been positive, but any feedback would be much appreciated
Please note that since this is a major update you will need to perform a settings reset using this file prior to installing the update: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/raw/V3.0-beta/settings-reset.uf2
This also necessitates re-pairing to all host devices
The default build of the firmware can be found here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/actions/runs/5158714204

gslavi, 2023-07-17:
I am having similar issues after updating to v3.0 today.  I was having a variety of issues with bluetooth and left to right half connectivity with v2.0.  After updating to v3.0, I was able to connect to my Dell Latitude 5430 running windows 10, iPad running the latest version of iPad OS (16.5.1), and a machine running Ubuntu 22.04.2 with no issues.  

When I try to connect to my 2016 MacBook Pro running Monterey I am unable to.  'Adv360 Pro' shows up as a device I can connect to within the bluetooth menu, but when I click on connect nothing happens.  When I click connect a second time the button momentarily grays out, only to come back seconds later and no connection between the two devices.  I am able to connect to other bluetooth mice and keyboards, but have not found a work around for the Adv360 Pro.  Is this a known issue? Anyone have similar problems or have a work around?

kylepairish, 2023-07-17:
> I am having similar issues after updating to v3.0 today.  I was having a variety of issues with bluetooth and left to right half connectivity with v2.0.  After updating to v3.0, I was able to [...]

Yes. I have this issue. Refer to https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/104

robertkluin, 2023-08-16:
I'm still having this issue with v3 firmware. I've got two Adv360 keyboards, one at home one at my office. I can pair and use one, then pair and use the other. But, once I attempt to connect to the first paired keyboard it fails to reconnect. They'll both gray out in the bluetooth menu. If I toggle Bluetooth on the mac it will usually reconnect with the v3 update. I don't have this issue with other bluetooth devices.

[image]

robertkluin, 2023-08-16:
Apple: 2021 - Apple M1 Pro

robertkluin, 2023-08-16:
Also, I'm on Ventura (Version 13.5 (22G74))

kylepairish, 2023-08-16:
This is the exact issue I’ve been having.

KinesisCorporation, 2023-10-20:
Please re-flash with today's update and try again.

We just re-tested connecting two 360 Pros to one Macbook Pro (2023 running 13.5.1) and had no issues. Both successfully paired and could send keystrokes to the Mac at the same time.

[image]

diorman, 2023-10-30:
Bluetooth connection works for me on a Macbook pro 2019 but does not work on a Macbook pro 2015. Both Macbooks have macos Monterey and the keyboard has been flashed including changes from 10/20/2023 😞

KinesisCorporation, 2023-10-30:
Hmm. So this is a different issue than the OP. They were trying to connect two keyboards to one Macbook. This is one keyboard to two Macbooks.

Curious if its an issue of it can connect to two at once, or something specific to the 2015 model? Can you try erasing the connection with both Macbook's at both the PC and keyboard end, and then try repairing the keyboard with the 2015 Macbook. Can you perform that test and then create a new issue describing the results?

KinesisCorporation, 2023-10-30:
@robertkluin @kylepairish @afilipchik 

Any update on the two keyboards to one Macbook issue? It _seems_ to have been resolved with the most recent update so will close this issue if we don't hear from you shortly.

ReFil, 2023-10-30:
2015 MacBook Pro might not support Bluetooth LE with the standard wireless card. You might need to use a Bluetooth LE supported USB dongle

diorman, 2023-10-30:
> Hmm. So this is a different issue than the OP. They were trying to connect two keyboards to one Macbook. This is one keyboard to two Macbooks.

@KinesisCorporation My bad! I'll create a new issue if there isn't an existing one related.

> Curious if its an issue of it can connect to two at once, or something specific to the 2015 model? 

It looks like it is specific to the 2016 model (I mistakenly typed 2015)

> Can you try erasing the connection with both Macbook's at both the PC and keyboard end, and then try repairing the keyboard with the 2015 Macbook. Can you perform that test and then create a new issue describing the results?

I already tried this multiple times without any luck

> 2015 MacBook Pro might not support Bluetooth LE with the standard wireless card. You might need to use a Bluetooth LE supported USB dongle

@ReFil the bluetooth version on the problematic macbook is 4.2 which I believe is compatible, insn't it?

Thanks!

kylepairish, 2023-10-31:
@KinesisCorporation I have been using the keyboard in wired mode for the last few months. I will update the firmware and try it again. Just for clarification my issue was that I was using 1 keyboard to connect to two different computers...1 is a macbook pro the other is my work windows laptop. Feel free to close and if the issue persists I will open a new issue specific to the issue I am having.

KinesisCorporation, 2023-10-31:
@kylepairish Thanks for the update. We know the keyboard can connect to two different PCs and toggle between them. If you are still having a problem please submit a trouble ticket directly to Kinesis and we'll help you out.

https://kinesis-ergo.com/support/contact-a-technician/

KinesisCorporation, 2023-10-31:
@diorman We just tested pairing directly with a mid 2015 MBP with Bluetooth 4.2 (no dongle) and it worked fine. And then we paired 2023 MBP and was able to switch between them.

ReFil, 2023-11-06:
Following up on this issue, is anyone still encountering this problem on the latest firmware update? If not I will close this issue out as solved in 48h

ReFil, 2023-11-08:
Closing issue as it seems to be solved, If people are still having connectivity issues please open a new GitHub issue for the specific problem you are encountering

closed 2023-11-08 by ReFil