KinesisCorporation/Adv360-Pro-ZMK #82: can't seem to switch default layers versus toggling

brokenalarms, 2022-12-20:
This may be a ZMK-specific question, but it seems like a pretty fundamental use case as a user going from previous Kinesis Advantage2 QD where I would swap back and forth from QWERTY->DVORAK multiple times a day, and still enjoy the same layer support.

How can I change the default layer with ZMK? It doesn't seem to implement the fairly fundamental eg what looks like `&df` in QMK. I simply want to go a DVORAK layer without the 'mod' light coming on and thereby disabling all other mod functionality (eg turning on keypad), because I am already considered to be in a 'mod' layer. Neither `&to` or `&tog` accomplish this. 

Where is the missing info here? Given this would be the first thing a lot of users do - simply replicate the QWERTY->DVORAK switching capability present on their old keyboard, but using ZMK, it would probably be a good idea to have a default 'Advantage 2QD' keymap download to get started here :) 

Thank you!

KinesisCorporation, 2022-12-20:
Hi Daniel, each key is individually re-assignable in every layer. There is no "quick config" for Dvorak at the moment but perhaps that is something we could create as a resource.

If you are primarily Dvorak, then you would probably want to reassign the primary 4 layers (Top, Keypad, Fn and Mod) to Dvorak and then create an additional QWERTY layer. If you are worried about the Layer LED then go deeper than layer 7 or 8 which have no color associated.

If you are primarily QWERTY, then you just need to add additional layer as Dvorak that you can jump into as needed.

Mod is just another layer and you can have Mod layer access keys in any of the other layers. Or you can put Mod commands in a typing layer on the hotkeys or other "available" keys.

As you can see, the possibilities are wide-ranging which is why we haven't tried to make a quick config as of yet.

zemackdaddy, 2022-12-21:
My Dvorak layer is the first and has a &tog to my second, qwerty layer.

Both layers seem fine using &mt and &tog to get at other layers that come later like Mod and Fn and others I've added. 

Is that what you're after?
Both base typing layers are still able to toggle or move to any other?

brokenalarms, 2022-12-21:
Thanks for the 7-8 tip, good to know. I'll try that and let you know!

ReFil, 2023-01-06:
Hi, just following up, has this been solved so i can close this issue out?

ReFil, 2023-01-12:
Closing as not planned

closed 2023-01-12 by ReFil