KinesisCorporation/Adv360-Pro-ZMK #83: Mod Morphs without keep_mods applies modifier to the first press of the modded binding

zemackdaddy, 2022-12-23:
Here is a [branch](https://github.com/zemackdaddy/Adv360-Pro-ZMK/tree/v2-plain) made directly from the v2.0 repo here.

In the most recent commit, you can see I added this behavior:

		pd3: prog_dvorak_rcurly_three_def {
			compatible = "zmk,behavior-mod-morph";
			label = "pd3";
			mods = <(MOD_LSFT|MOD_RSFT)>;
			#masked_mods = <(MOD_LSFT|MOD_RSFT)>;
			#binding-cells = <0>;
			bindings = <&kp RBRC>, <&kp N3>;
		};

and bound it to the 4th key (3) in the top left of the left half.

The output without shift is as expected:  }

### _The Problem_
The output when holding shift is not.
The first time 3 is pressed while holding shift:  #
Subsequent presses while *still holding the same modifier*:  3

Only 3 should be output by this mod morph when holding shift.

It doesn't matter how long you hold shift before the first press of the key.  It is always #.  All further presses are 3s (until you let go of the shift).

I raised this in the zmk discord in the help-forum and bravekarma  was kind enough to attempt reproduction on what they said was [basically zmkfirmware/main](https://github.com/zemackdaddy/Adv360-Pro-ZMK/blob/v2-mouse-keys/config/boards/arm/adv360/adv360.dtsi).  They were not able to duplicate.  only a 3 is output while holding shift.

The suggestion is that this might be fixed if the [refil/zmk/adv360-z3](https://github.com/ReFil/zmk/tree/adv360-z3) was brought up to match zmkfirmware/main.

The [mod morph](https://zmk.dev/docs/behaviors/mod-morph) should not pass the modifiers on unless keep_mods is specified.

Mod Morphs in general are working.  I guess the theory is that something has been fixed since refil/main was forked.

ReFil, 2022-12-23:
Will investigate, thanks for bringing to my attention

eckig, 2022-12-29:
I have a similar problem with this mod morph in my config: 

```
rkp_n7: right_parenthesis_kp_n7 {
        compatible = "zmk,behavior-mod-morph";
        label = "right_parenthesis_kp_n7";
        #binding-cells = <0>;
        bindings = <&kp DE_RPRN>, <&kp N7>;
        mods = <(MOD_LSFT|MOD_RSFT)>;
    };
```

Normal press (`)`) works fine but with Shift (`7`) it only works part of the time printing nothing.
When I started using this config, the feature was still a PR but working fine. Only updating to the current V2.0 branch "broke" this recently.

ReFil, 2023-01-12:
The base zmk branch has just been updated with the latest changes from zmk main, are you ok to rebuild fw and try again?

eckig, 2023-01-12:
I just rerun the workflow file and flashed both sides (had to do the reset first) - the issue is still present.

zemackdaddy, 2023-01-12:
I will test hopefully later today tomorrow and let you know (I'm OP).

zemackdaddy, 2023-01-13:
It still doesn't work for me.

I built locally with Docker and was careful to --no-cache.

ReFil, 2023-01-14:
did the build process you used rerun `west update`? nothing in the branch modifies those behaviours so now it's updated i'm not sure why there would be issues

eckig, 2023-01-15:
So I tested using simple buttons like `&kp KP_N7` and guess what: Does not work either - looks like my key is dead and I need to contact kinesis.

zemackdaddy, 2023-01-15:
@ReFil I used this [branch](https://github.com/zemackdaddy/Adv360-Pro-ZMK/tree/v2-plain).

It has two changes past the upstream Kinesis branch:

- the mod morph in the keymap
- Makefile uses a --no-cache

KinesisCorporation, 2023-01-15:
> So I tested using simple buttons like `&kp KP_N7` and guess what: Does not work either - looks like my key is dead and I need to contact kinesis.

It's certainly possible you have a bad key switch.  Which switch are we talking about? Before we issue an RMA though we'd want you to install a copy of the factory default firmware which you can download here: 

https://kinesis-ergo.com/support/kb360pro/#firmware-updates

eckig, 2023-01-16:
> It's certainly possible you have a bad key switch. Which switch are we talking about? Before we issue an RMA though we'd want you to install a copy of the factory default firmware which you can download here:
> 
> https://kinesis-ergo.com/support/kb360pro/#firmware-updates
Tried that. The settings have been applied, but the Key does not output `7` on key press.

KinesisCorporation, 2023-01-16:
Roger that. Please submit a ticket and we'll issue an RMA and then replace the switch under warranty. Sorry for the inconvenience. 

https://kinesis-ergo.com/support/contact-a-technician/
https://kinesis-ergo.com/support/warranty/

yanshay, 2023-02-26:
I have a similar issue if not exactly the same. Please see zmkfirmware/zmk/issues/1685. Seems like it is specific to Kinesis Pro and it really bugs me. 
Any ideas what to do about it?

yanshay, 2023-03-04:
@zemackdaddy I noticed that in my case this happens whem karabiner elements is processing keyboard events (even if nothing to do on karabiner side). Once I disabled karabiner processing of the Kinesis keyboard this got resolve.
I wonder, are you using karabiner on Mac?

zemackdaddy, 2023-03-04:
> @zemackdaddy I noticed that in my case this happens whem karabiner elements is processing keyboard events (even if nothing to do on karabiner side). Once I disabled karabiner processing of the Kinesis keyboard this got resolve.
> I wonder, are you using karabiner on Mac?

@ReFil the problem occurs on Mac with and without Karabiner and also on Windows.

007kevin, 2023-04-26:
my [mod-morph](https://github.com/007kevin/Adv360-Pro-ZMK/blob/v2.4/modspace/config/adv360.keymap#L35-L41) behavior is also sending the mod to the morphed key. investigation shows this issue is [fixed](https://github.com/zmkfirmware/zmk/pull/1412) but the latest doesn't seem to be included in the Adv360 builds. Is it possible to update the zmk?

ReFil, 2023-04-26:
https://github.com/ReFil/zmk/commit/ef2e6e9156806760d9e91c0d635cd809058c3ee7

That fix is merged into the branch the config repo builds from. I'm not sure why the fix isn't working on this application

zemackdaddy, 2023-05-29:
This is now working for me.

closed 2023-05-29 by zemackdaddy