KinesisCorporation/Adv360-Pro-ZMK #87: Binding Mouse Clicks/Actions as described in Manual

hh, 2023-01-04:
I'd like to use mouse clicks and actions as described in the manual, but I'm unable to find the key(mouse?) codes:

From Page 13 of the manual:
> 5.2 Four new HotKeys

> These four keys can be programmed to perform any single key action, mouse click, macro and more.

From Page 17 of the manual:

> 6.4 ZMK Token List

> ZMK supports a wide array of keyboard actions (letters, numbers, symbols, media, mouse actions). Visit the link below for a handy list of tokens to reference when programming your keyboard.
https://zmk.dev/docs/codes/

hh, 2023-01-04:
There is a reddit discussion, but would love to hear from Kinesis on this as it's in the manual but possible not supported by the Adv360-Pro-ZMK yet. 

https://www.reddit.com/r/ErgoMechKeyboards/comments/w39hpt/zmk_users_how_to_add_mouse_keys/

KinesisCorporation, 2023-01-05:
That was our mistake in the manual. It was copied over from the SmartSet model keyboard which supports mouse clicks and movements.

One of our power users has built an experimental branch of ZMK firmware for the 360 that has Mouse Keys supports. It doesn't have the indicator lighting features and its still experimental, but feel free to take a look.

 https://github.com/urob/adv360-demo-config

ReFil, 2023-01-06:
Closing as resolved, Mouse key support will be added to this repo when it's merged into ZMK main

closed 2023-01-06 by ReFil

sudoFerraz, 2023-08-02:
@ReFil any updates on such support?

ReFil, 2023-08-02:
Mouse support still hasn't been merged into zmk main unfortunately

sudoFerraz, 2024-11-05:
@ReFil Any plans to add an official beta branch with mouse support from ZMK?