KinesisCorporation/Adv360-Pro-ZMK #90: Build failure on MacOS using podman

humblehacker, 2023-01-06:
```
Error: preparing container d3ed7931ce28a2b3f8dccb090151f120c89435662bf43e646923c88f7c167449 for attach: lsetxattr Adv360-Pro-ZMK/config/adv360_left.keymap: operation not supported
```

The changes in PR #58 are the culprit. After reverting locally, I'm able to build successfully.

For reference: https://github.com/containers/podman-compose/issues/509

ReFil, 2023-01-12:
Would you be able to test my PR? I dont have a mac to test on

kibner, 2023-01-13:
> Would you be able to test my PR? I dont have a mac to test on

I have an M2 MBA on Ventura 13.1 and the PR still doesn't work. I ran `make clean` and then `make`. This is my error:

```
STEP 9/9: CMD ["./build.sh"]
COMMIT zmk
--> deed133ce29
Successfully tagged localhost/zmk:latest
deed133ce29b5c4f550a8a7dd2c2279c77b14f51865d30a4f28b72b67b0c5918
/opt/homebrew/bin/podman run --rm -it --name zmk \
                -v /Users/kibner/src/Adv360-Pro-ZMK/firmware:/app/firmware:z \
                -v /Users/kibner/src/Adv360-Pro-ZMK/config:/app/config:ro,z \
                -e TIMESTAMP=20230113202756 \
                zmk
Error: preparing container 24c739ca68cf1ba4724e869dc56491774750e777a4542628153483c298ebfa34 for attach: lsetxattr /Users/kibner/src/Adv360-Pro-ZMK/firmware: operation not supported
make: *** [all] Error 126
```

ReFil, 2023-01-13:
can you put "uname" into terminal and post what you get from the terminal please?

kibner, 2023-01-13:
> can you put "uname" into terminal and post what you get from the terminal please?

```
kibner@Kibners-MacBook-Air Adv360-Pro-ZMK % uname
Darwin
```

ReFil, 2023-01-13:
Thanks for doing that for me. i'm trying to have it detect the OS and remove those SELinux params appropriately, i'll fiddle a bit more and let you know

ReFil, 2023-01-13:
Please try it again, the uname command adds a sneaky bit of whitespace in that has to be explicitly removed

kibner, 2023-01-13:
> Please try it again, the uname command adds a sneaky bit of whitespace in that has to be explicitly removed

It works!

ReFil, 2023-01-18:
@humblehacker have you had the opportunity to test the changes i've made? it'd be much appreciated if you were able to confirm whether or not they solve the issue

humblehacker, 2023-01-25:
@ReFil Sorry for the late reply, and thanks for the fix. I checked out your branch and was able to successfully build!

PR #91 (2023-01-25): Disable SELinux support on OS X
Uses uname to classify the operating system and disables the SELinux 'z' flag if that's determined to be the case, should resolve #90
closed 2023-01-25 by ReFil