KinesisCorporation/Adv360-Pro-ZMK #97: Is it possible to switch layout with different profiles?

mertzjames, 2023-01-30:
I have a Mac, Windows, and Linux machine that I'm connecting to my Kinesis Advantage 360 Pro to.  I was wondering if there was a way to switch profiles and layouts for each of those.  I really liked the ability to switch the Windows, Command, and Home keys on my old Advantage 2.

ReFil, 2023-01-30:
You could use a macro to execute the layer swap and profile swap with one keypress

itsthejoker, 2023-03-06:
I wrote my own macro for this -- you can take a look at my repo for how to do it yourself. https://github.com/itsthejoker/Adv360-Pro-ZMK

KinesisCorporation, 2023-03-10:
@mertzjames closing the issue for now. Please reopen if you aren't able to get it to work.

@itsthejoker thanks for sharing!!!

closed 2023-03-10 by KinesisCorporation

itsthejoker, 2023-03-29:
@mertzjames just fyi I updated my repo with the Mac layer to the latest V2 firmware, since my fork was previously the day 1 release.

mertzjames, 2023-04-10:
@itsthejoker I'm just now seeing this.  Sorry.  It's been a busy couple of weeks/months for me. Thanks for putting this together.  I'm going to go over it later today but do you have a quick and dirty overview of what I should look at?  I'm still pretty new to ZMK and the layouts.  Is there documentation of the different key codes and how to use Macros to switch layouts?

itsthejoker, 2024-10-14:
@mertzjames FYI I finally got off my ass and updated my repo to v3. If you just want to install my version, then Mod+M will enable mac-specific keys. https://github.com/itsthejoker/Adv360-Pro-ZMK