KinesisCorporation/Adv360-Pro-ZMK #102: Add french keymap

boris-sbizzera, 2023-02-13:
It seams that i can't use azerty default keymap (french layout).
any suggestions?
thank you

KinesisCorporation, 2023-02-14:
Hi Boris, 

The Advantage360 is a not a standard keyboard layout so its not as simple as just enabling AZERTY in your OS. But you can re-assign keys to perform any supported action in any layer.

boris-sbizzera, 2023-02-14:
so how can i assign for example the key "é" to be on Num_2 space on querty?
Thank you

KinesisCorporation, 2023-02-14:
Here is the list of supported key codes

boris-sbizzera, 2023-02-14:
so I can't create an azerty layout?

boris-sbizzera, 2023-02-14:
I can't add this layout

[image]

ReFil, 2023-02-14:
You should set it up as a normal qwerty then configure your operating system in azerty mode

boris-sbizzera, 2023-02-15:
thanks for your help!

closed 2023-02-15 by boris-sbizzera

sieman, 2023-03-17:
Thanks for this issue. I'm looking for a solution for a german layout (especially neo-layout.org).