KinesisCorporation/Adv360-Pro-ZMK #104: My Adv360 Pro cannot connect to my mac studio。

xbear23, 2023-02-16:
My Adv360 Pro cannot connect to my Mac Studio via Bluetooth, but can connect to the left keyboard via USB. I've tried all the methods but can't connect. Has anyone else encountered the same situation?

ReFil, 2023-02-16:
Have you ensured the Bluetooth profile is clear? Can you go into specifics of what you've tried?

billjive, 2023-02-17:
This happens to me on a daily basis. I can get it paired to my MacBook Pro but after longer periods of idle, or if I take my laptop somewhere else and bring it back it almost always nukes the connection. At a minimum I need to clear the BT profile on the keyboard, forget the device on my Mac and reconnect to the "new" device in the BT list.

xbear23, 2023-02-20:
Actually, when I first got the keyboard, it was able to connect to my Mac Studio. However, one day it suddenly disconnected and hasn't been able to connect again since then.
It can now connect to my other MacBook M1 and even to my iPhone, but it just can't connect to my Mac Studio, which is the primary computer I use for work.
I have tried the following methods:

1. Disconnecting the keyboard's Bluetooth connection on the Studio and attempting to reconnect.
2. Resetting the Bluetooth settings by pressing the "Mod+Command" keys on the keyboard and attempting to reconnect.
3. Attempting to reconnect by pressing "Mod+1~5" on the keyboard.
4. Deleting the keyboard on the Studio and attempting to reconnect.
5. Updating the official firmware： “Adv360-Pro-Factory-Default-Firmware-v2.0-11-1-22”.

 After I deleted the keyboard from the Studio for the first time, it couldn't be connected again. My keyboard can be found by the Studio, but when I click on the "Connect" button, a loading icon briefly appears, then the "Disconnect" button is displayed, which immediately changes back to the "Connect" button. No matter how many times I repeat this process, it doesn't work.

I even asked ZMK on Discord and the response I got was "Several people have asked the same question" and then they sent me this link：https://zmk.dev/docs/troubleshooting#split-keyboard-halves-unable-to-pair, But I haven't tried it yet because I'm not sure if it's suitable for my situation and I'm afraid it might cause more unnecessary trouble.Thus, I am now choosing to ask you first.

Also, I can now use my Adv360 Pro like a wired keyboard by connecting the left side of the keyboard using a USB cable.

xbear23, 2023-02-20:
I have resolved the issue by repeatedly pressing "Mod+Windows" several times. Thank you very much, you can close this issue now.

billjive, 2023-02-20:
Um, weird. xbear23 had a really detailed post that doesn’t appear here any more (I got the notification in email). His experience is EXACTLY mine. I have also tried Mod+Windows several times and it doesn’t help the situation.

I’ve started to just plug in my keyboard via a USB cable just so it reliably works.

I’m going to quote the post xbear23 made here because seriously it’s the same experience I’ve had:

> I have tried the following methods:
> 
> Disconnecting the keyboard's Bluetooth connection on the Studio and attempting to reconnect.
> Resetting the Bluetooth settings by pressing the "Mod+Command" keys on the keyboard and attempting to reconnect.
> Attempting to reconnect by pressing "Mod+1~5" on the keyboard.
> Deleting the keyboard on the Studio and attempting to reconnect.
> Updating the official firmware： “Adv360-Pro-Factory-Default-Firmware-v2.0-11-1-22”.
> After I deleted the keyboard from the Studio for the first time, it couldn't be connected again. My keyboard can be found by the Studio, but when I click on the "Connect" button, a loading icon briefly appears, then the "Disconnect" button is displayed, which immediately changes back to the "Connect" button. No matter how many times I repeat this process, it doesn't work.

xbear23, 2023-02-20:
> 

I had actually tried many times before to clear the keyboard's Bluetooth settings using "Mod+Windows(Command)", but only pressed it once. Today, after submitting my post, I tried again and pressed it many times, including on the left and right sides. Then, my keyboard successfully connected, so I deleted my post. Now, everything is working fine with my keyboard.

KinesisCorporation, 2023-02-20:
Bill-

Is your Profile LED flashing slow or fast? Slow indicates there is still a pairing stored on the keyboard, fast would indicate that the keyboard itself is ready to be paired. If its flashing slow then it hasn't actually been cleared.

Is it possible your "BT Clear" command got moved/deleted from the Mod layer on your keyboard? Or more likely you moved the Windows key in your primary layout" and thus Mod + Windows isn't actually triggering a command at all.

xbear23, 2023-02-21:
In my experience, even if the profile LED is fast flashing, it may not necessarily connect successfully. Before, I was certain that my profile LED was flashing quickly, and I tried switching profiles many times using "Mod + 1~5", but it didn't work. So, try pressing "Mod+Windows" multiple times, including in combinations on the left and right sides, because that's what ultimately worked for me to successfully connect my keyboard. It's really strange, though.

ReFil, 2023-02-22:
If someone could get logs from the mac bluetooth handler that may prove helpful in answering why this is happening

ReFil, 2023-02-23:
What version of MacOS are you using? did this appear after an update?

billjive, 2023-02-24:
This has happened me since Day 1 using the keyboard. It's always been with MacOS Ventura, any version has the same issue.

The LED flashes fast and re-pressing Mod + Windows multiple times doesn't change anything.

What's the best way to get the logs you want? I could record anything related to bluetooth in the MacOS console app (logs)..

ReFil, 2023-02-24:
Have you updated the firmware on your board at all? Sometimes finicky bluetooth issues can be fixed by flashing the settings reset file available in this repository to both halves then loading the latest firmware. Note that this will unpair it from the computer and it will need to be "forgot" in macos before connecting again as the encryption keys need to be regenerated

Thanks for getting logs, it would be useful to see why the mac drops the connection. You can either attach them here as a txt file or email them to me at willow@polarityworks.com

billjive, 2023-02-24:
Yep I updated to the 2.0 firmware from the GitHub repo back when I got the keyboard.

I'll see if I can capture logs next time this happens...

ReFil, 2023-02-24:
Could you also try installing the settings reset firmware as detailed in this link: https://kinesis-ergo.com/support/kb360pro/#troubleshooting

please use this settings reset file: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/raw/V2.0/settings-reset.uf2

And this firmware: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/suites/10964917033/artifacts/571668418

billjive, 2023-02-24:
Ok - done. I'll let you know if that makes the bluetooth connection more reliable.

Now my new problems is I can't remember how I got my MacOS mappings working correctly on this keyboard. Do you guys have any pointers?

KinesisCorporation, 2023-03-10:
@billjive What do you mean by MacOS "mappings". Do you mean just changing your thumb keys to the historical Advantage "Mac mode"?

arogulin, 2023-03-22:
I have exact same issue as @billjive: when I get my MBP16 to another room for a call and then go back, sometimes (not every time, I would say 50/50) the keyboard starts to slowly flash the blue LED (profile 2). And much more rarely (maybe once in 2 weeks) this happens after the weekend when I have not used this macbook, but used the keyboard with another macbook. I keep a wired keyboard for this all the time on my table to be able to enter a password and the procedure to fix this issue is the following:
1. Press Bluetooth resetting shortcut on the keyboard, which in my case is Mod+Alt(left thumb cluster, the top right key) to put it into pairing mode.
2. Open Bluetooth settings in MacOS (Ventura 13.2.1) and forget "Adv 360".
3. Open any other settings in MacOS and then return back to the Bluetooth tab, "Adv 360" will appear on the bottom, available for connection.
This procedure works like a charm every time, but it's not the best user experience to keep a spare keyboard on the table, or when I'm in a hurry and every second counts.

I will wait for @billjive to report his results on flashing the settings reset firmware, and if that will help - I will do the same. I was blaming MacOS for this issue, but looks like it could be the keyboard itself.

billjive, 2023-03-22:
I was waiting to comment to have my keyboard operate for a couple weeks with the original firmware. I can say it's definitely working better. At this point I'm using the firmware and settings reset suggested above in this thread.

So by working better, what I mean is that more consistently I can come back to my computer after it sleeps, wake it up, and the keyboard will pair and be fine. What I notice however consistently is if I undock my computer from my monitor and travel around with it, then come back, the keyboard never pairs. I have to do the usual forget-keyboard dance and repair it from scratch. (basically what @arogulin is doing).. so @arogulin my keyboard is working better but I'm seeing the same behavior as you...

arogulin, 2023-03-22:
@billjive I will give it a try then, thank you for replying so quickly. I just flashed the `settings-reset.uf2` on both sides and then flashed my own firmware fork with all the updates from the upstream (as of 2023-03-06). Not sure if I needed to flash the stock firmware before my own. I will report here about the results once I have them.

arogulin, 2023-03-25:
Just experienced the issue again after taking the laptop to another room for about 30 minutes, so can confirm that the `settings-reset.uf2` workaround does not help with that issue, unfortunately.

KinesisCorporation, 2023-03-29:
Thanks for the additional information. We apologize for the inconvenience. Our team is aware of these issues and we are working on a solution but we don't have any specific time table.

ReFil, 2023-04-12:
Someone has reported a workaround that might help get the connections to the succeed on the latest Ventura.

* Open up Keychain Access
* Removing "passwords" related to the MAC address of the keyboard
* Rebooting and re-pairing

arogulin, 2023-04-14:
@ReFil I just had the issue again this morning. I will try this workaround, though my guess is that it won't make a difference. I checked the Keychain Access and I have exactly 1 "password" for my keyboard's MAC address and it has "last modified" date of this morning when I re-paired the keyboard. If I delete that record, then reboot, then re-pair the keyboard, it will probably create exact same record with new encryption key, so the end result will be the same as before the workaround. Not sure how that may help, but any workaround is worth a try.

ReFil, 2023-04-14:
Ok thanks, if you have a fork of the config repository there's also some items you can enter into the left _defconfig file that might help, if this doesn't work I'll give you those params too to try

kelvinlauKL, 2023-04-17:
> After I deleted the keyboard from the Studio for the first time, it couldn't be connected again. My keyboard can be found by the Studio, but when I click on the "Connect" button, a loading icon [...]

I'm having the exact same issue!

kelvinlauKL, 2023-04-23:
I'm not sure how to find the MAC address of the keyboard. Google / ChatGPT mentioned it could be in the bluetooth settings, but I'm unable to see it.
[image]

kelvinlauKL, 2023-04-25:
After days of using wired connection, I managed to reconnect to my laptop. Some things I noticed:

1. Happened when I took my laptop outside (disconnected due to out-of-range)
2. I managed to do a fresh pair after removing all the bluetooth devices in my second MacBook - I needed to do this for the two times I lost connection

Given point 2, I suspect it has something to do with bluetooth interference. That being said, I don't use many bluetooth devices - (keyboard, mouse, trackpad, AirPods Pro, AirPods Max).

ReFil, 2023-04-25:
Interesting, it's unlikely to be interference on its own but it's possible that resets something in the Bluetooth stack on the Mac

ReFil, 2023-04-26:
After investigating with the Apple Accessory Design guidelines, a potential fix has been found

Can someone encountering this issue please add these lines to the bottom of adv360_left_defconfig
```
CONFIG_BT_PERIPHERAL_PREF_TIMEOUT=600
CONFIG_BT_PERIPHERAL_PREF_MIN_INT=12
CONFIG_BT_PERIPHERAL_PREF_MAX_INT=24
CONFIG_BT_PERIPHERAL_PREF_LATENCY=30
```
Then remove the line for NKRO (Line 53)
```
CONFIG_ZMK_HID_REPORT_TYPE_NKRO=y
```
and rebuild firmware. In my testing on an M1 MBP it was able to repeatedly connect successfully

ReFil, 2023-05-02:
Just following up, have any of the people experimenting with this change have noted any difference in the Bluetooth reliability with MacOS?

billjive, 2023-05-02:
I have built it but haven't installed. I've noticed more stability with Mac OS 13.3.1 so I'm going to continue with that for another week or so.. if it starts to fail earlier then I'll install that fix.

billjive, 2023-05-08:
I've been using this for about a week and haven't seen any change. In general, the stability of the BT connection with my Mac is better (something I attribute to the latest OS version) but when I take my laptop away for a while and come back to my desk the keyboard will not reconnect. I have to go through the same steps of forgetting the device and re-pairing.

cvcore, 2023-05-18:
> I've been using this for about a week and haven't seen any change. In general, the stability of the BT connection with my Mac is better (something I attribute to the latest OS version) but when I [...]

I'm having the exact same issue.

metalra1n, 2023-05-25:
same issue as well. M1 Air on Ventura. If I take my laptop out of the room for abit and come back in the keyboard will not reconnect to the laptop. Best thing I have found is to open the laptop and turn bluetooth on and off and that gets the keyboard to reconnect.

metalra1n, 2023-05-30:
> 

this may have been addressed in 13.4. So far so good, will update shortly.

kelvinlauKL, 2023-05-31:
I went on a trip recently. Made sure to power off my keyboards before I took my MacBook out of range. Two weeks later, I turned on the keyboard and it connected without a hitch. 

That's my solution for now I guess

ReFil, 2023-06-02:
We've developed a new major update for the Adv360 Pro with significant improvements to the bluetooth code especially with regards to mac compatibility and split connection reliability.
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/tree/V3.0-beta
It would be very helpful if people encountering issues with macs and split connectivity could test this branch. Our initial testing on our devices has so far been positive, but any feedback would be much appreciated
Please note that since this is a major update you will need to perform a settings reset using this file prior to installing the update: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/raw/V3.0-beta/settings-reset.uf2
This also necessitates re-pairing to all host devices
The default build of the firmware can be found here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/actions/runs/5158714204

metalra1n, 2023-06-05:
Pretty positive my bluetooth issue with waking the mac up fro sleep is fixed with this 3.0 update. Will follow up again to confirm tomorrow morning.

metalra1n, 2023-06-06:
> Pretty positive my bluetooth issue with waking the mac up fro sleep is fixed with this 3.0 update. Will follow up again to confirm tomorrow morning.

Unfortuneatly, this issue still exists. If the laptop is taken out of the room overnight and I bring it back in the morning and try to wake it from sleep with the keyboard it will not connect to the keyboard. My logitech bluetooth mouse is able to wake it from sleep though and then I have to open the laptop from clamshell mode and toggle bluetooth off and on in order for the keyboard to connect to the laptop again.

ReFil, 2023-06-06:
What mouse are you using? does the issue present if the laptop is kept in the room with the keyboard overnight?

metalra1n, 2023-06-06:
> What mouse are you using? does the issue present if the laptop is kept in the room with the keyboard overnight?

Logitech MX Master 3. No, the issue only happens if I take the laptop away from the keyboard overnight.

mpereira, 2023-06-21:
One datapoint:

I'm on macOS and have two macbooks paired with my adv360 pro. I had a V1.0-based firmware installed.

My right split stopped working a couple of days ago. It would either have the three leds blinking red or no lights on. It wouldn't work even when connected via USB.

I first tried reinstalling my V1.0-based firmware. That didn't change anything. Then I tried to update to a V2.0-based firmware. That also didn't fix the issue.

Then I followed [@ReFil's instructions to get on the V3.0 beta](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/104#issuecomment-1574187079). My right split started working afterwards, connected via USB and bluetooth.

Thanks @ReFil and crew for the work towards V3.0!

mpereira, 2023-06-21:
Two additional notes:

1. I also experience macbook wake ups when the keyboard is connected. I'll report back if I notice changes now that I'm on V3.0.
2. It would be really nice to have keymaps defined in a way that doesn't conflict with firmware updates, as [others have mentioned](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/167).

billjive, 2023-06-23:
I'm still using the firmware I installed on May 2nd (see thread above) and am seeing much better results with the latest MacOS updates. Must have been some updates in the OS that are also helping here. I haven't installed the 3.0 firmware version yet, but wanted to mention that the version I'm using is better .. finally!

KinesisCorporation, 2023-06-23:
That's also great to hear. Can you confirm the MacOS version you are currently on.

metalra1n, 2023-06-28:
The keyboard now is doing the same thing with Windows devices as well. If the computer goes to sleep and the keyboard is taken away from the device it will not reconnect via bluetooth until the bluetooth is cycled on the computer itself. 

Scenario 1:
Keyboard is connected to laptop and working. Put computer to sleep. Turn keyboard off. Turn keyboard on and try to wake computer from sleep. Fails to connect. Logitech MX Master 3 connects no issue. In order for keyboard to reconnect I need to turn bluetooth off andn then on again. Keyboard instantly connects.

Scenario 2:
Keyboard connect to MacOS and working via bluetooth. I take laptop out of the room and use it while in bed. I go to sleep closing the lid of macbook. Next morning I take laptop into office and plug into external monitor. I try to wake computer from sleep using keyboard, it fails to connect. I then have to use the mouse (Logitech MX Master 3) to wake mac from sleep. Keyboard still does not work. Have to open laptop from clamshell mode and type in password with on-device keyboard. I then have to turn bluetooth off and then on again in order for the keyboard to connect.

This occurs on MacOS and Windows. I am using 3.0 firmware. 

Please let me know if I can provide anymore information that would be helpful.

billjive, 2023-06-28:
> That's also great to hear. Can you confirm the MacOS version you are currently on.

Ventura 13.4

KinesisCorporation, 2023-06-28:
@metalra1n 

Thanks for the report.

Scenario 1:
- Are you using the Logitech MX Master 3 over BT or with the "Logi Bolt" dongle?
- What about if you don't turn off the 360 keyboard and let it go to sleep automatically?

Scenario 2: 
- This is with a different 360 keyboard at the office right?

metalra1n, 2023-06-28:
> @metalra1n
> 
> Thanks for the report.
> 
> Scenario 1:
> 
> * Are you using the Logitech MX Master 3 over BT or with the "Logi Bolt" dongle?
I am using the mouse with BT. No dongle.

> * What about if you don't turn off the 360 keyboard and let it go to sleep automatically?
Same issue. 
> 
> Scenario 2:
> 
> * This is with a different 360 keyboard at the office right?
No, this is the same keyboard. The only time I turn it off with the on/off switch is when traveling to and from the office.

ReFil, 2023-06-28:
Enabling USB logging could provide more insight into why it's failing to reconnect, I'll send over some instructions shortly. To clarify, was it previously able to wake the windows pc from sleep and then just stopped working one day?

metalra1n, 2023-07-06:
> To enable usb logging follow the section here: https://zmk.dev/blog/2022/04/02/zephyr-3-0 on enabling the CDC device and pointing to it in the chosen node in adv360.dtsi
> 
> Then add CONFIG_ZMK_USB_LOGGING=y to the adv360_left_defconfig file and rebuild

I have logging configured. What would you like me to do next?

ReFil, 2023-07-07:
If you connect it to a different computer than the one it is failing to connect to over Bluetooth, you can monitor the logs as it attempts to connect. If you add &out OUT_BT it will send keystrokes over Bluetooth too. The logging mode causes the device to enumerate as a USB to serial device. Logs can be retrieved with any program like termite, putty or tera term.

kylepairish, 2023-07-07:
> If you connect it to a different computer than the one it is failing to connect to over Bluetooth, you can monitor the logs as it attempts to connect. If you add &out OUT_BT it will send keystrokes [...]

This morning it did it again. I came in and plugged my laptop into the monitor then woke it from sleep using the logitech mouse. The keyboard would not connect via bluetooth and when I open the bluetooth menu on MacOS the advance360 device is greyed out. The keyboard bluetooth light is blinking slowly but does not connect to the computer. I was watching the logs after I plugged the keyboard back in via usb and I dont see anything showing up related to it trying to connect to the device. It clear's the ring buffer and then just sits there while the bluetooth light blinks slowly. If I turn the bluetooth on and off on the laptop it instantly reconnects and I see the pairing messages in the log. 

I apologize, where do I add the &out OUT_BT in the firmware? I tried finding it in the zmk docs but didn't get anywhere.

kylepairish, 2023-07-07:
This log is when switching the bluetooth profile to my windows work laptop. The interesting thing here is if I unplug the keyboard from USB (its currently connected to my Mac and logging) and try to use the keyboard which is "supposedly" connected to the windows PC via bluetooth...it doesnt work even though it says its connected in Windows. Again, the log below is from switching from bluetooth profile 0 (macOS) to profile 1 (windows work laptop). 

[17:48:04.938,415] <dbg> zmk: zmk_keymap_apply_position_state: layer: 3 position: 2, binding name: BLUETOOTH
[17:48:04.938,476] <dbg> zmk: zmk_ble_prof_select: profile 1
[17:48:04.938,537] <dbg> zmk: update_advertising: advertising from 2 to 0
[17:48:04.938,751] <dbg> zmk: zmk_usb_get_conn_state: state: 3
[17:48:04.938,751] <dbg> zmk: get_selected_transport: Both endpoint transports are ready. Using 0
[17:48:04.973,388] <dbg> zmk: split_central_notify_func: [NOTIFICATION] data 0x2001043b length 16
[17:48:04.973,846] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 7
[17:48:04.973,876] <dbg> zmk: position_state_changed_listener: 7 bubble (no undecided hold_tap active)
[17:48:04.973,937] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 7, binding name: MO
[17:48:04.973,968] <dbg> zmk: mo_keymap_binding_released: position 7 layer 3
[17:48:04.973,999] <dbg> zmk: set_layer_state: layer_changed: layer 3 state 0
[17:48:05.014,221] <dbg> zmk: kscan_matrix_read: Sending event at 4,2 state off
[17:48:05.014,373] <dbg> zmk: zmk_kscan_process_msgq: Row: 4, col: 2, position: 2, pressed: false
[17:48:05.014,404] <dbg> zmk: position_state_changed_listener: 2 bubble (no undecided hold_tap active)
[17:48:05.014,465] <dbg> zmk: zmk_keymap_apply_position_state: layer: 3 position: 2, binding name: BLUETOOTH
[17:48:05.018,981] <dbg> zmk: split_bt_update_led_payload:

This log is when I switch the profile back to profile 0 (trying to connect to MacOS):

[17:49:05.010,467] <dbg> zmk: zmk_keymap_apply_position_state: layer: 3 position: 1, binding name: BLUETOOTH
[17:49:05.010,528] <dbg> zmk: zmk_ble_prof_select: profile 0
[17:49:05.010,589] <dbg> zmk: update_advertising: advertising from 0 to 2
[17:49:05.011,352] <dbg> zmk: zmk_usb_get_conn_state: state: 3
[17:49:05.011,383] <dbg> zmk: get_selected_transport: Only USB is ready.
[17:49:05.161,254] <dbg> zmk: kscan_matrix_read: Sending event at 4,1 state off
[17:49:05.161,437] <dbg> zmk: zmk_kscan_process_msgq: Row: 4, col: 1, position: 1, pressed: false
[17:49:05.161,468] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
[17:49:05.161,529] <dbg> zmk: zmk_keymap_apply_position_state: layer: 3 position: 1, binding name: BLUETOOTH
[17:49:05.183,288] <dbg> zmk: split_central_notify_func: [NOTIFICATION] data 0x2001043b length 16
[17:49:05.183,746] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 7
[17:49:05.183,807] <dbg> zmk: position_state_changed_listener: 7 bubble (no undecided hold_tap active)
[17:49:05.183,837] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 7, binding name: MO
[17:49:05.183,898] <dbg> zmk: mo_keymap_binding_released: position 7 layer 3
[17:49:05.183,929] <dbg> zmk: set_layer_state: layer_changed: layer 3 state 0
[17:49:05.206,329] <dbg> zmk: split_bt_update_led_payload: 

The bluetooth connection is currently not working on either laptop. I paired them both last night and had them both working but after coming back into the office this morning neither will connect even though on the windows laptop it states that its connected and on MacOS the Advance360 Pro device under bluetooth is greyed out even when the keyboard is trying to pair with it...

Hopefully this is helpful.

ReFil, 2023-07-17:
The keyboard will always try to connect to any paired devices and will shows connected to any paired devices even if the profile is not selected. The profile selection chooses where the keystrokes go. Unfortunately these logs don't offer any insight into what the cause of the issue is, I will do further investigation

pgeadas, 2023-08-07:
Hi. Was having no issues on version 2 up until recently when I could no longer reconnect. Tried version 3 and is even worse. It connects but it is constantly disconnecting so it is unusable as I cannot type a full sentence. Went back to v2 with full reset and everything and I was able to connect again. However, today morning I could not connect again and had to do the same thing which is boring. I'm on a Mac and I just closed the lod yesterday as usual, nothing else. I'm on MacOS Ventura 13.5.

KinesisCorporation, 2023-08-18:
Hmm... It could be Ventura 13.5 issues (although its working fine for us here). Or perhaps you didn't use the 3.0 Settings Reset file before upgrading to 3.0 firmware the first time. Using the wrong reset file seems to cause some weird failure modes from what we've seen.

If you're up for it, we'd love to have you flash both sides with the 3.0 Reset and then install the default 3.0 firmware from the link below and report back.

https://kinesis-ergo.com/support/kb360pro/#firmware-updates

pgeadas, 2023-08-18:
I did reset with with the 3.0 version and cleared with the respective
version beforehand twice actually. And it presented the same issue both
times. I even restarted the Mac in between just to make sure and nothing
seemed to work. So I'm just glad that in the end the version 2.0 worked
again... Isn't there I way to see any logs or something to debug what is
happening with v3.0?

kelvinlauKL, 2023-09-18:
Well, I shut down my laptop for the first time in months. Keyboard is having trouble connecting again.

moritzschatz, 2023-09-19:
Mabook Pro M1 (2021), MacOS 13.5.2, newest stable 3.0 firmware

When i boot the Macbook the light is flashing slowly, not able to enter my password. after entering my password via the build in Keyboard and logging in the light turns on constant and the bluetooth connection is established.

[43 more replies]

closed 2024-04-15 by ReFil