KinesisCorporation/Adv360-Pro-ZMK #108: Profile LED Stays Green After Checking Battery Levels

patricksalo, 2023-02-23:
**Prerequisite**
Use profile #5 which keeps the profile (person icon on left keyboard half) off when paired to a device

**Steps To Reproduce**
1. Press Mod+3 (left half) or Mod+4 (right half) to display battery levels via the LEDs
2. After checking battery level via LEDs, release the keys from step 1
3. All the LEDs on both halves turn off, except the profile LED remains green

**Work Around to Turn Off LED**
1. Change your profile to one of the LED lighted ones, 1-4, (e.g. Mod+#) then switch back to the unlit profile 5 (e.g. Mod+5)
2. The profile light will now be turned off

**Unknowns**
1. My current battery level is >80% so the LED color for step 3 above may or may not always be green
2. Potentially the profile LED color might correspond to one of the other battery level colors such as yellow, orange or red, but at the moment, I can't really determine this

ReFil, 2023-02-23:
Thanks for the bug report, this is really in depth, it's very much appreciated. I will investigate ASAP

ReFil, 2023-02-24:
This should be fixed with this https://github.com/ReFil/zmk/commit/566212be98baeaf57c26ae56bb24641e08909560

Please rerun build step and let me know

patricksalo, 2023-02-25:
This worked great. Many thanks.

closed 2023-02-25 by patricksalo