KinesisCorporation/Adv360-Pro-ZMK #110: Right half doesn't reconnect if keyboard is shut off while in bluetooth profile other than 1

FlippingBinary, 2023-02-24:
This may be related to #65 and I posted a comment in that thread to describe a set of repeatable steps, which I've updated here based on another experience:

1. Press [Mod] 3 (to switch to a Bluetooth profile other than 1).
2. Count to 20 (or 10 if you're impatient).
3. Flip the power switch of each module on simultaneously.
4. Observe the right module's three LEDs blinking red.

I had initially thought this was some how associated with Bluetooth profiles that are undefined, but it happens in profile 2 for me as well, which is connected to my phone. After several attempts to get them to reconnect with synchronized power cycles, I was able to get them to reconnect through some unusual steps with my phone. I'm not sure which part caused them to reconnect with each other, but here they are:

1. Disable Bluetooth on phone.
2. Wait 10 seconds.
3. Enable Bluetooth on phone and select the keyboard to connect.
4. Wait for connection to fail.
5. Power cycle both keyboards at the same time.
6. Watch keyboard connect automatically via Bluetooth with both modules communicating again.

The alternative would be to press both reset buttons simultaneously. That works as well, and it's what I used when it was stuck in an unpaired Bluetooth profile.

I have no trouble turning them to reconnect if they're in profile 1.

Can someone else confirm if they are able to repeat this experience?

ReFil, 2023-02-24:
Are you running out the box firmware or have you updated to the latest?

FlippingBinary, 2023-02-24:
> Are you running out the box firmware or have you updated to the latest?

I think I remember it being tricky to get them to connect on the out-of-the-box firmware, but I have been running the latest firmware for a few months. As of today, it was eight commits behind but the problem persisted after updating it.

However, I noticed the hard reset instructions in the user manual and decided to give that a try. Now the modules seem to connect to each other easily. When I read those instructions before, I had assumed that process was for users who modify their firmware beyond key bindings, but maybe it should be the first step before updating the firmware the first time.

ReFil, 2023-02-24:
Flashing the settings reset firmware and updating to the latest base firmware (by rerunning your build action in github or make clean locally) will often help with connection instability

KinesisCorporation, 2023-03-17:
Thanks for sharing this. We tested several times and it didn't fail and we had a response all ready to go... and then one time it did and we could not get the keyboard to resync. It _seems_ like it might be a specific timing of the power-cycle.

To restore the connection we had success disabling the PC Bluetooth before powering up the keyboard. Not ideal, but a decently quick workaround. Can you confirm?

We'll investigate and its very possible this will be resolved with the Direct Advertising update that is already  in the works. 

If anyone can boil the repro steps down even further, please do let us know.

cc @ReFil

brokenalarms, 2023-03-23:
Yeah it still happened for me after a fresh flash. Was very frustrating multiple resets of each half, and the mashing Mod+1 made no difference either. If I deleted the Bluetooth profile on my Mac, it would boot up ok, allow pairing... Then when you choose pair, it will pair (adds keyboard to list of connected devices) but typing doesn't work. Then switch off and on and you'll find it still unpaired again, with the right side flashing again.

I'm not sure how I got it to work but I think that yes it might actually have been removing the device, switching off Bluetooth and restarting with Bluetooth off before re-pairing.

itsthejoker, 2023-03-31:
I just hit this myself while troubleshooting a bluetooth connection issue w/ support, and I can confirm that hitting the reset button on both sides at the same time does let them reconnect (though sometimes it does take a few tries).

brokenalarms, 2023-04-03:
Tried that many times and didn't fix it for me. I had to reset the image.

brokenalarms, 2023-04-12:
any updates on this issue? I had used the keyboard since reflashing for about a week (judging from issue last lodged), now have the same thing again. Suddenly stopped working on all 3 computers its synced to, despite zero changes.

If I switch BT off and on while the keyboard is off, then turn the keyboard on, I get this instead:
[image]

And generally when I try to click 'connect' to either keyboard, it loads, connects, then the button just reappears with 'connect' again and the keyboard continues flashing in pairing mode.

brokenalarms, 2023-04-12:
https://user-images.githubusercontent.com/7388170/231345542-ee43fcb5-71fb-49c5-9446-fe93b93f1c53.mov

ReFil, 2023-04-12:
Is the keyboard is pairing mode (fast profile led flashing) or just disconnected (slow profile led flashing)?

KinesisCorporation, 2023-04-12:
@brokenalarms 

It sounds like we're at the point where we should swap out your keyboard for a new one so we can try to understand whether this is a ZMK problem, a your-setup problem, or a one-off hardware failure.

Please submit a trouble ticket here and provide link to this thread.
https://kinesis-ergo.com/support/contact-a-technician/

brokenalarms, 2023-04-12:
actually, it was disconnected. Confusing since the device still appears as a nearby unconnected device with a 'connect' option when not in pairing mode, and when paired leaves this list once it is 'properly' connected as expected.

Given that this happens every few weeks, I'll take the advice and submit a ticket thanks @KinesisCorporation .

moon0326, 2023-04-28:
Leaving my experience here. 

I also had a Bluetooth connectivity issue with my second Mac. It wouldn't connect, so I removed Adv360 from the Bluetooth list by clicking `Forget`

[image]

After doing so, Adv360 appeared under `Nearby Devices`, but clicking `Connect` didn't work. It briefly connected and then disconnected immediately, similar to what happened in brokenalarms's [video](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/110#issuecomment-1504555273).

What I did not know was the difference between `paring` and `disconnected` mode (which I learned from @ReFil 's comment). My keyboard was discoverable, but it was in `disconnected` mode. 

What worked for me was resetting my Bluetooth 2 by pressing Mod+Right Window key. This problem is constantly reproducible in my case.

So if you're experiencing a similar issue, try resetting your connection to see if that works.

I really should have read the official manual carefully.

> 5.7 Bluetooth Clear
If you wish to re-pair one of the 5 Bluetooth Profiles with a new device (or are having trouble connecting to the
current device), use the Bluetooth Clear shortcut (Mod + Right Windows) to erase the connection with PC for
the current Profile. If you are simply trying to re-pair with the same device we recommend disconnecting/
removing the “Adv360 Pro” from the target PC and performing the Bluetooth Clear command for a clean slate.

Matt-Herndon, 2023-04-28:
I think this may be happening with my keyboard as well. I was just experiencing fast red flashing on the right board. I did quick succession resets left then right by using paper clips to try and fix the issue because this had worked in the past. Sadly it did not work this time after multiple attempts. Going into my mac and explicitly forgetting the keyboard did allow a new quick reset to work though. Now the keyboard is working with its left side plugged into my computer and the right side i being wireless.

FlippingBinary, 2023-04-30:
Just chiming back in to say that I haven't had any of these problems since flashing the settings reset firmware and updating to the latest base firmware, as described [above](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/110#issuecomment-1443930464). I highly recommend it to anyone who is experiencing intermittent connectivity issues.

brokenalarms, 2023-05-03:
This was happening for me on the latest firmware, and @KinesisCorporation had also reproduced on the latest. So still very much alive. Logging back in to report it happened again this morning - right keyboard flashing, no idea how to fix, restarted keyboard multiple times, tried plugged in, etc. Fortunately this time I had the computer I'd previously been using available (on profile 1), and it seemed that if I started it up with that computer on, it got unstuck and I could at least pair it again to the current computer (on profile 2) - but I needed to first remove/forget/cycle bluetooth and re-add the device in order for it to work.

FlippingBinary, 2023-05-03:
@brokenalarms I wasn't trying to suggest the problem doesn't exist for anyone anymore, only that the critical step for me was to flash the settings reset firmware. It is a special firmware that only resets the internal settings so a regular firmware will have a fresh slate when it is flashed over the settings reset firmware. I had previously thought that flashing the latest firmware would be the most thorough way to reset the keyboard, but it wasn't. If you have only been flashing regular firmware, I highly recommend giving the settings reset firmware a try before flashing the latest firmware. 

Here are the current instructions, according to the [support website](https://kinesis-ergo.com/support/kb360pro/):

**Settings Reset** – If you are having trouble with your keyboard you can perform a “Settings Reset” to return the keyboard to its native state, and then install a fresh copy of firmware.

1. Download the “Settings Reset” and Default Firmware Files below (3 total .uf2 files).
2. Follow the instructions in the guide at left but install the settings-reset.uf2 file on BOTH the left and right “ADV360PRO” drives. _Note: The keyboard will be disabled once reset until new firmware is installed._
3. Then follow the instructions in the same guide to install the left and right side firmware files. _Note: After firmware installation the left and right modules should re-sync with each other automatically. If  they do not, try power-cycling them at the same time._

* **[Settings Reset](https://kinesis-ergo.com/wp-content/uploads/settings_reset-11-1-22.zip)**
* **[Factory Default Firmware (v2.0.116 3-9-23)](https://kinesis-ergo.com/wp-content/uploads/Adv360-Pro-Factory-Firmware-3-9-23-for-upload.zip)**

KinesisCorporation, 2023-05-03:
Hi folks,

We've just posted a "core ZMK 3.2" firmware file for the Adv360 Pro. This has none of the custom code that was developed specifically for the 360 (eg RGB indicators).

For those still having connectivity problems, this would be a great file to install and test drive for a while to see if persistent issues are resolved or not. And then report back here with the results.

***You must flash the Settings Reset file on BOTH modules before installing this firmware since its a "MAJOR" update***

https://kinesis-ergo.com/download/advantage360-professional-core-zmk-3-2/

cvcore, 2023-05-08:
Hi @KinesisCorporation , I have the same issue as OP and have already tried the `ZMK3.2-vanilla` firmware in https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/110#issuecomment-1533761857, but unfortunately I can still observe the old behavior, with the right keyboard accidentally failing to register with the left one during poweron.

However, I can observe when there are multiple bluetooth devices nearby the keyboard - in my case two laptops and several phones - the chance of the registration failure increases. So it seems the keyboard is somehow very sensitive to its neighboring devices.

One explanation of this behavior could be due to `CONFIG_BT_MAX_CONN` (with default value `5` for LH module, see [4]) restricting the maximum connection the bluetooth module could establish (see ref. [1]-[3] for more info on bluetooth behavior of zmk). So when there are multiple neighboring devices during keyboard power-on, the LH keyboard may accidentally pick already 6 devices other than the RH module, blocking it from connection.

So as a temporary solution, I changed the settings to
```
CONFIG_BT_MAX_CONN=16
CONFIG_BT_MAX_PAIRED=16
```
in my `configs/arm/adv360/adv360_left_config`, and the problem seems to be gone now. Note the RH module setting should be kept default (set to `1` for both variables).

Maybe you can try this as well @FlippingBinary and let me know if it works.

[1] https://zmk.dev/docs/behaviors/bluetooth
[2] https://docs.zephyrproject.org/3.0.0/reference/kconfig/CONFIG_BT_MAX_CONN.html?highlight=config_bt_max_conn
[3] https://docs.zephyrproject.org/3.0.0/reference/kconfig/CONFIG_BT_MAX_PAIRED.html#std-kconfig-CONFIG_BT_MAX_PAIRED
[4] https://zmk.dev/docs/config/system

brokenalarms, 2023-05-08:
> @cvcore <https://github.com/cvcore> that seems like a good theory. I do
> have more than five Bluetooth devices in my office, so it's reasonable to
> believe that they were interfering. However, I haven't had trouble since I
> flashed the firmware reset image and then flashed the latest firmware. If
> the keyboard starts to act up, I will try your suggestion.

Hmm with mine, I can get it unstuck by restarting the laptop that it was
last paired to (1), then switching to the new device I was trying to pair
in the first place (2). However, I still need to delete the device from 2
and re-pair on 2 every time otherwise it just flashes slowly (paired but
can't find device). Also, I need to have the device last connected to
handy, which is a fairly major requirement as I used to leave it at work
sometimes. Otherwise it's a reflash if I want to use it that day 😔

andrewlmurray, 2023-05-20:
Hello - I ran into this issue tonight. Found this ticket while looking for a solution. I have only two profiles I use - a mac on profile 1 and my windows pc on profile 2. 

My repro case was very simple:

1. I was on profile #2 connected to my PC. 
2. I restarted my PC. 
3. Upon restart my keyboard did not re-pair so I couldn't enter my password. 
4. I turned off both halves of the keyboard.
5. I turn on both halves of the keyboard. The left side paired to the PC, the right side showed three flashing lights. 

To attempt to fix I tried the steps listed in the FAQ. (power cycled, reset left and then right at almost the exact same time) Those steps did not help.

I also rebased my firmware fork against the latest from this repo, (my changes are only to my keymap) and reflashed both halves. Still would not connect. 

I then tried the trick that @FlippingBinary listed in his issue. I tried connecting w/ my phone saw it fail and then power-cycled. Both halves then started talking happily again. 

I've used your keyboards forever and love my 360. Hope you can find a fix for this...

andrewlmurray, 2023-05-20:
After reading through the responses to this ticket - just want to add some details in case it is helpful. 

1. I was on a semi-modern firmware when this happened. (sometime late march was the last time I updated)
2. I do not have a plethora of bluetooth devices in range. Just my laptop, phone and my PC.

brokenalarms, 2023-05-22:
> ...I wonder if the devices that you have connected in specific slots are a
> clue? I also have a Mac in slot 1 and a PC in slot 2. 🧐

Both my devices are Mac so don't think it makes a difference there..

ReFil, 2023-06-02:
We've developed a new major update for the Adv360 Pro with significant improvements to the bluetooth code especially with regards to mac compatibility and split connection reliability.
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/tree/V3.0-beta
It would be very helpful if people encountering issues with macs and split connectivity could test this branch. Our initial testing on our devices has so far been positive, but any feedback would be much appreciated
Please note that since this is a major update you will need to perform a settings reset using this file prior to installing the update: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/raw/V3.0-beta/settings-reset.uf2
This also necessitates re-pairing to all host devices
The default build of the firmware can be found here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/actions/runs/5158714204

cvcore, 2023-06-16:
Hi @ReFil , thanks for the update and effort spent in investigating the issue. 

Here are my observations after having upgraded to the firmware built with the V3.0-beta branch, tested on a model with serial no. starting with 360PRO310** (purchased in Mar. 2023): 

- Switching to a bluetooth profile other than 1 before powering off no longer disturbs the split connection during power-on (as in https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/110#issue-1597834499).
- I don't observe refusal of connection as in https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/110#issuecomment-1504555273
- When powering on the keyboard, split disconnectivity can still be observed from time to time. But in most cases can be resolved just power-cycling the left part. Ideally I don't expect this to happen at all, so this is still something left for optimization.

ReFil, 2023-07-23:
Can anyone replicate this issue on V3.0 firmware?

FlippingBinary, 2023-07-23:
> Can anyone replicate this issue on V3.0 firmware?

I don't usually turn it off and on, so I tested it before writing this comment. I turned both halves off, waited a few seconds, then turned them both on at the same time. The right half failed to connect to the left, but the left one connected to the computer. I power cycled a few more times, alternating between a slight lead on the left one or the right. At one point, the connection indicator on the left was solid white, indicating a connection to the computer, but the computer just said it was just paired, not connected. I power cycled one more time and now I can type again.

There's a possibility my computer could be causing problems -- it is the other link in the chain, after all. So again with it connected to my laptop on profile 2. After power cycling, the left half showed a white connection light instead of blue and I could not get the right half connected until I pressed the reset button on both halves simultaneously. At one point, the left half was unresponsive with no lights even though the switch was in the on position.

Since I don't usually travel with the keyboard, the more troubling problems for me lately have been stuck keys and the connection dropping. 

The stuck key seems to get unstuck if I disable Bluetooth on the computer and re-enable it. I used to try power cycling it to get the key unstuck, but that would often trigger the difficulty in getting the right connected to the left. There didn't seem to be any pattern to the occurrence of stuck keys. They computer just seemed to think a key was held, and no other key presses would register.

When the Bluetooth connection fails to work, I just bypass Bluetooth with a USB cable. Disabling and re-enabling Bluetooth on the computer doesn't fix it and power cycling doesn't fix it. The keyboard was getting unreliable, so I flashed the v3 settings reset and firmware again (even though it was already on v3) a few days ago and haven't had that problem since then, but the right half still struggles to connect with the left.

ReFil, 2023-07-25:
Hmm I see, I'm trying to separate different problems out so I can target specific problems with bugfixes. I'm not sure that your current connection troubles are related to this issue. Could you open a new issue for your stuck key problem?

ReFil, 2023-10-07:
Given there haven't been reports of this specific profile connection issue occuring on v3.0 I'm closing this issue as solved

closed 2023-10-07 by ReFil

FlippingBinary, 2023-10-26:
This problem continues to persist. I'm not sure what else I can do about it. I even bought a different Bluetooth dongle just to make sure it wasn't the Bluetooth card in my desktop causing issues. The stuck key problem seems to occur when the keyboard disconnects from Bluetooth. I assume Windows repeats the key on its own until it receives a keyup event from the keyboard, or until it recognizes that the keyboard is disconnected/unresponsive. Is there some kind of diagnostic mode I can use to record logs so I can send you something next time it happens? The only way I can be sure I'm not interrupted by this problem is by leaving it plugged in, but then I don't notice how frequently it disconnects from Bluetooth until the right one needs to charge and I forget to switch it back.

ReFil, 2023-10-26:
Are you using the latest firmware? It brings in some changes to host connections. I cannot replicate this behaviour on my setup since the first v3.0 update. You can enable logging but it interferes with Bluetooth and sleep mode so it would probably not be too helpful in this case

KinesisCorporation, 2023-10-26:
Hey Jon, we get this is really frustrating.

If you would please use the attached instructions to install a 2.0 AND 3.0 Reset, and then flash the 10-23-23 default firmware (please download all files from the link below). These instructions are designed to eliminate all the weird potential complications that be introduced during an update (its overkill 99% of the time) so follow them to the letter. If its still not working then we definitely want to issue an RMA and either get you a new Pro or switch you into the USB model, whatever you prefer.

https://kinesis-ergo.com/support/kb360pro/#firmware-updates

[](url)
[Advantage360 Professional - Firmware Update Instructions 10.25.23 (KB360-PRO-GBR).pdf](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/13181176/Advantage360.Professional.-.Firmware.Update.Instructions.10.25.23.KB360-PRO-GBR.pdf)

FlippingBinary, 2023-10-26:
Thank you! I will try your suggestions tomorrow. I was already running the 3.0 beta that was posted in this thread, but I will try the two different resets before trying the 10-23-23 default firmware. I'm just happy to have something new to try.

ReFil, 2023-11-06:
Hi Jon, has the newest update shown any change in behaviour?

rpisarew, 2023-11-06:
For me now it is better, but I still have sometimes to disable/enable my BT to get a connection when my MacBook was not in range of the keyboard for a longer time. And it takes long to get a connection on a Windows PC.

Rarely I got repeating or delayed keys, but disable/enable BT helps to resolve that..

ReFil, 2023-11-06:
Thank you for the feedback, this GitHub issue specifically pertains to a problem where the splits will not reconnect if turned off in a different Bluetooth profile, for some of your other Bluetooth problems there's other issues that have been opened. People are currently experimenting with changing the BLE parameters which seems to improve the connection reliability e.g. https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/259#issuecomment-1794396919

FlippingBinary, 2023-11-06:
@ReFil My two halves seem to connect reliably if the left half is first either plugged in or its Bluetooth profile LED is solid. Then the right half has no trouble connecting. However, if I turn the right half on at the same time as the left, or before the left has turned on its Bluetooth profile LED, the right half won't connect and the left half fails to connect to Bluetooth. I don't know if that is the expected behavior, but this is perfectly acceptable to me because I can reliably get it to connect.

I apologize for conflating the connection problem between the modules with the connection problem between the keyboard and the computer. The initially described problem is resolved, so I'll close this issue. Thank you for your patience and for the link to the other issue. It looks like it might solve the Bluetooth connection problem I've been having.

ReFil, 2023-11-06:
Hi Jon

No worries, glad to be of assistance. The behaviour you're describing with the split connection is expected right now, I'm going to investigate to see if we can improve the experience agnostic of power on order. If you're still having connection issues to your Mac after trying  some of those other fixes feel free to open another issue, we're trying our best here to ensure every 360 Pro user gets the best experience!

brokenalarms, 2023-11-06:
Just on this, if you're supposed to have the left on first, then you should
amend your manuals too, as I repeatedly read instructions to turn both on
at the same time. Cheers!