KinesisCorporation/Adv360-Pro-ZMK #140: Build failing using make (windows using WSL2 and docker)

Arethusag, 2023-04-16:
Running the make command gives the following error:

`/usr/bin/env: 'bash\r': No such file or directory
make: *** [Makefile:16: all] Error 127`

ReFil, 2023-04-16:
I've followed all the setup commands as per the readme and i cannot replicate this issue on WSL2 Ubuntu 20.04. What WSL distro are you using?

Arethusag, 2023-04-16:
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.1 LTS"
PRETTY_NAME="Ubuntu 22.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.1 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

Error seems to be coming from docker run command in the makefile. docker build command is working fine.

```
/usr/bin/docker run --rm -it --name zmk \
        -v /mnt/c/*REPOPATH/Adv360-Pro-ZMK/firmware:/app/firmware:z \
        -v /mnt/c/*REPOPATH*/Adv360-Pro-ZMK/config:/app/config:ro,z \
        -e TIMESTAMP=20230416161412 \
        zmk
```

ReFil, 2023-04-16:
the SELinux variables should not be causing a problem, it's only for Macs as they do not support docker SELinux

ReFil, 2023-04-16:
I suspect windows style line endings is configured on git, these will add extra characters into the files that the WSL subsystem cannot handle

ReFil, 2023-04-16:
A solution would be to run these commands in the folder where the adv360 files are contained
`sudo apt-get install -y dos2unix # Installs dos2unix Linux
sudo find . -type f -exec dos2unix {} \; # recursively removes windows related stuff`

Arethusag, 2023-04-16:
This resolved the issue with make. Also, cloning the repo right from WSL circumvented the issue entirely.

> A solution would be to run these commands in the folder where the adv360 files are contained `sudo apt-get install -y dos2unix # Installs dos2unix Linux` `sudo find . -type f -exec dos2unix {} \; # recursively removes windows related stuff`

Thank-you!

closed 2023-04-16 by ReFil