KinesisCorporation/Adv360-Pro-ZMK #143: CONFIG_ZMK_RGB_UNDERGLOW_ON_START does not work.

moon0326, 2023-04-20:
Hi,

I'm trying to turn off the led indicators on startup. 

Looking at the ZMK documentation, `CONFIG_ZMK_RGB_UNDERGLOW_ON_START` is the correct value.

I tried setting `CONFIG_ZMK_RGB_UNDERGLOW_ON_START` to `n`, but it did not work.

Am I changing the correct config value for Adv 360?

ReFil, 2023-04-20:
This is only for the RGB behaviour when the board is first initialised, after that the lighting values are stored in the flash memory. If you want the lighting to be off all you need to do is use the RGB toggle key command in the manual, then wait for a minute for the changes to be saved to flash, after that powering on and off the keyboard halves the lights should remain off

moon0326, 2023-04-20:
> 

Thanks. It still turns on when I turn off and on the keyboard halves.

Just to clarify, I'm trying to turn off the 6 led indicator, not the backlights.

ReFil, 2023-04-20:
how long did you leave it after turning off the indicator lights? There's a period in which changes are not saved to the flash memory in order to prevent flash wear

moon0326, 2023-04-20:
> how long did you leave it after turning off the indicator lights? There's a period in which changes are not saved to the flash memory in order to prevent flash wear

I made sure to wait for 5 mins. The led indicator under the user icon turns on.

ReFil, 2023-04-20:
I've investigated and found the bug causing the behaviour you mention, it will be patched soon

moon0326, 2023-04-27:
Thank you for the quick fix!

It's working well 👍

closed 2023-04-28 by ReFil

hwangdylan, 2023-05-02:
@ReFil how can I get the latest updates into my keyboard? I tried to rebuild my fork and reflashed but it didn't include the newest changes from your commit. Matter of fact I can't find that commit in the official zmk firmware. What am i missing here?

hwangdylan, 2023-05-02:
oh so i can see that when i build I get the latest commit but it seems like the light still turns on after a reboot. From the above comments it seems like i do not need to make any changes to CONFIG_ZMK_RGB_UNDERGLOW_ON_START and it should still work

moon0326, 2023-09-17:
Hi @ReFil 

This issue is occurring in v3 version again.

I flashed V3 version last week and noticed the same issue again.

Steps to reproduce:

1. Turn on the indicator (in my case, it's the 2nd light on the left half)
2. Turn it off by MOD + backspace
3. Wait for 5 mins
4. Turn off the both halves
5. Turn on the both halves
6. The led indicator is on again

ReFil, 2023-09-18:
That's really strange because the fix was brought forwards to v3, I will investigate further when i have the time

moon0326, 2023-10-03:
@ReFil Any updates?

ReFil, 2023-10-05:
i'm sorry i cant replicate this, the config option works properly for me using the latest version of this config repository

moon0326, 2023-10-06:
@ReFil Thanks for looking into it.

It now works as expected after flashing `Adv360 Pro Settings Reset Flash Erase` firmware.