KinesisCorporation/Adv360-Pro-ZMK #148: Turn off Bluetooth

sieman, 2023-04-27:
I using the Advantage360 Professional via USB wires and i have turn off bluetooth in MacOs. The Keyboard works well but the Bluetooth is still active on the keyboard. How to deactivate the bluetooth on the keyboard?

ReFil, 2023-04-27:
The bluetooth has to be active at all times because it's how the halves communicate

sieman, 2023-04-27:
Thanks for the answer. Then should the halves communicate via usb. Maybe it is some day possible.

closed 2023-04-27 by ReFil

ReFil, 2023-08-02:
It always prioritises the USB connection so even if the pairing slot is empty and someone pairs no data can be retrieved. Furthermore it's not pairable once the slot has been connected to. There is a passkey pairing mode for setups that require more security, it can be enabled in the config

ReFil, 2023-08-02:
As soon as usb connection is re-established data goes out that way. There's a visual indication of Bluetooth connection through the status LEDs. It's also not possible to brute force the passkey as it's entered dynamically on the keys during the pairing sequence and will timeout. 

Disabling Bluetooth will disable the connection between the halves unfortunately. There's no scope to have Bluetooth available only between the two halves within the current code framework. There is a pull request currently open pending testing in the base adv360z3.2 repo to disable Bluetooth advertising under certain circumstances

sieman, 2023-11-16:
[image]

I did a RF measurement with [The AC2 Acousticom](https://www.emfields-solutions.com/products/ac2-acousticom-2.asp) so in the picture you can see the result. 

Hopefully you find a solution to synchronise the two halfes via usb (and turn off Bluetooth).

aaronjensen, 2024-01-30:
@ReFil did the PR for disabling advertising ever get merged? I'm in a similar situation as OP and would like to turn BT advertising off entirely.

ReFil, 2024-01-31:
Hi @aaronjensen  unfortunately it didnt, it hooked into quite a few core functions and would have made maintaining the 360 Pro quite tricky. I might look into doing it at some point in a way that is easily maintainable