KinesisCorporation/Adv360-Pro-ZMK #151: lighting values not stored in the flash memory and reset after reboot

hwangdylan, 2023-05-02:
This is a follow up to this issue since I was unable to get it working even after the patch. I've commented additional information in the closed ticket linked below but just wanted to create a new ticket to track. 

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/143

hwangdylan, 2023-05-02:
The keyboard backlight saves but not the 6 indicator lights.

ReFil, 2023-05-02:
Have you tried the CONFIG_ON_START option? The initial state of the LEDs is always overridden by the start value for reliability purposes

hwangdylan, 2023-05-02:
Where can I find documentation on that? I can't seem to find it in the ZMK docs from my quick search. Also based on the other issue, it seems like that configuration wasn't required? Out of curiosity why is it needed now?

ReFil, 2023-05-02:
https://zmk.dev/docs/config/underglow

The documentation on the config options relating to RGB lights, including the ```CONFIG_ZMK_RGB_UNDERGLOW_ON_START``` option that governs the initial behaviour of the lighting can be found here. It's worth noting that due to the custom features the ```CONFIG_ZMK_RGB_UNDERGLOW_EFF_START``` option has a higher range to allow for the extra lighting selections to be enumerated

The firmware had a bug where it would ignore the ON_START value and any saved indicator state in the flash. It has since been corrected to take ON_START into account

hwangdylan, 2023-05-02:
Ahh okay thanks! That makes a lot more sense. So I just saw you updated the other comment in the other issue with : 

> this isn't quite accurate, brightness and "effect" are saved but on state is overriden

Why is it being overridden? Because in the normal zmk doc it states 

> The *_START settings only determine the initial backlight state. Any changes you make with the [backlight behavior](https://zmk.dev/docs/behaviors/backlight) are saved to flash after a one minute delay and will be used after that.

What was the reasoning behind having to override this feature?

ReFil, 2023-05-02:
Since the lighting on the Adv360 Pro is used for displaying useful information it was deemed better to avoid any conflicts with other features that modify the flash saved RGB state such as the idle timeout in favour of ensuring the lights will always behave reliably in a manner specified by the user in the ON_START value, as these other features were causing inconsistent properties to be saved to the flash and thus unpredictable RGB states when the board was rebooted

closed 2023-05-02 by ReFil