KinesisCorporation/Adv360-Pro-ZMK #158: Hyper key (question)

agzam, 2023-05-21:
Can someone please help me, I can't figure out how to make a hyper key.

I need a key that emulates holding Ctrl+Alt+Cmd+Shift. I'd like to bind it to a single key or to a Fn+Key. I got "sticky" behavior, but that's not exactly what I want.

agzam, 2023-05-21:
I was able to add hyper by combining `LS(LC(LA(LGUI)))`. I assigned it to a single key. Unfortunately, my attempts to bind it to a key-combo were not very successful. It seems I need to tweak `tapping-term-ms`, `quick_tap_ms`, etc., but there's currently no way (as it seems) to configure them through the editor and I'm hesitant of directly tweaking them, since all that would be overriden if I mess with something in the editor.

ReFil, 2023-05-22:
I dont think the editor overrides those parameters once configured, are you ok to try doing it manually and seeing what happens?

ReFil, 2023-05-30:
Just following up, has this issue been solved?

agzam, 2023-05-30:
I didn't have time to play with it, but I will soon. Closing this for now. Thank you for your assistance.

closed 2023-05-30 by agzam