KinesisCorporation/Adv360-Pro-ZMK #160: Bluetooth disconnects from device 1 frequently when I use device 2 (via USB-C)

moon0326, 2023-05-24:
Hi,

So I started using my Adv 360 with a computer that is in a separate room. I always connect the Adv 360 with a USB-C cable when using it with the computer in the room, as I experience Bluetooth issues whenever I connect more than one device to the Adv 360, just like many other people on GitHub and Reddit.

When I take the keyboard back to the living room to use it with Device 1, it disconnects 2-3 times daily. I have to remove the Bluetooth profile from Device 1, reset the connection on the keyboard, and then reconnect all over again.

I have already tried all the suggestions from other users who faced similar issues.

Is there anything else I can do, apart from always connecting via USB-C?

I hate to say this, but the more I use Adv 360 with multiple devices, the more I think only the Bluetooht 1 connection is usuable. There are so many outstanding Bluetooth issues on Github and Reddit since the launch. Most people simply gave up at this point.

ReFil, 2023-05-24:
What device is connected on profile 1? Is it a Mac or a windows PC, have you tried using it on another profile to confirm the issue isn't specific to profile 1? Can you go into more detail of the suggestions you've tried?

moon0326, 2023-05-24:
> Is it a Mac or a windows PC

Device 1: Mac Studio with Mac OS Monterey (12.3)
Device 2: Macbook Pro with Mac OS Ventura (latest)

> have you tried using it on another profile to confirm the issue isn't specific to profile 1

I haven't. I can try that, but I really don't expect it to be the problem. I don't have any problem if I don't use Device 2 for a few days. It's only when I connect to Device 2 and then switch back to the Device 1. 

> Can you go into more detail of the suggestions you've tried?

I initially had Bluetooth issues described in https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/110 and https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/81

The solutions suggested in the issues did not work. I thought they worked at first, but I was wrong.

ReFil, 2023-05-24:
When you say "Device 1" and "Device 2" Are these corresponding to the bluetooth profile slots? apologies but i'm getting confused when you refer to devices, profiles and bluetooth 1 connection. Please try the ZMK 3.2 firmware and let me know how that affects behaviour.

moon0326, 2023-05-24:
> When you say "Device 1" and "Device 2" Are these corresponding to the bluetooth profile slots? apologies but i'm getting confused when you refer to devices, profiles and bluetooth 1 connection. Please try the ZMK 3.2 firmware and let me know how that affects behaviour.

My apologies.

I used to use both Bluetooth profiles 1 and 2, but gave up using profile 2.

I'm not using:

Device 1 with Bluetooth profile 1
Device 2 with USB-C 

> Please try the ZMK 3.2 firmware and let me know how that affects behaviour.

Thanks, but..I'm just going to give up and use USB-C for both devices. I've already spent enough time trying to fix these connectivity issues and now I have to try a brand new firmware without even knowing the cause. I hope Kinesis finds a solution to these Bluetooth issues. I'll check back in a few months..

closed 2023-05-24 by moon0326

cvcore, 2023-06-17:
Sorry for commenting on the closed issue, but ultimately Kinesis should provide the user an option to just connect to the active device via bluetooth. 

The current behavior of connecting to all paired devices only cause trouble to me in most cases. The scenario I can imagine simultaneous connection could be useful is to have automatic switching of bluetooth profile when seeing multiple monitors from multiple devices. But I also don't see this feature advertised.

Related issue: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/172