KinesisCorporation/Adv360-Pro-ZMK #164: V3.0 doesn’t work.

ebzzry, 2023-06-03:
The branch V3.0-beta doesn’t work at all with me. After flashing the keyboard, the middle left LED on the left half would keep blinking, white the right half doesn’t have any blinking LEDs, at all.

I reverted back to the V2.0 branch.

ReFil, 2023-06-03:
Did you perform a settings reset prior to upgrading? The lights on the right side should slow flash red until connecting. Middle led on left side flashing is normal behaviour. Did you try pressing the mod key (top left key on right side board)? This should illuminate an led green on both sides

ebzzry, 2023-06-03:
Yes, I did all of that.

ReFil, 2023-06-03:
Does the green led illuminate on both sides when you hold the mod key down?

ebzzry, 2023-06-03:
Prior to flashing, yes. After flashing, not anymore.

The procedure that I used was the same one that I have been using with V2.0.

ReFil, 2023-06-03:
If you connect the left side to the computer over USB do keys work?

ebzzry, 2023-06-04:
They don’t completely work. So, what I did to reflash it back to V2.0 was to use the pins.

ReFil, 2023-06-04:
Are you using the default keymap? Did you use the settings reset file in the V3.0 beta branch as it was updated as part of the upgrade?

michaelglass, 2023-06-05:
I had @ebzzry's experience until I used [the reset file.](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/raw/V3.0-beta/settings-reset.uf2)

ebzzry, 2023-06-05:
Let me do that.

ReFil, 2023-06-05:
Just following up, @ebzzry have you had the opportunity to do this yet?

ReFil, 2023-06-07:
Has resetting with the correct file yielded any change?

closed 2023-06-08 by ebzzry