KinesisCorporation/Adv360-Pro-ZMK #167: Merge conflict with new firmware update

Valentin-Seehausen, 2023-06-07:
Hey,

when I try to merge your firmware update, I get merge conflicts.

I would not like to re-create all my key mappings. Also, I don't know how to resolve the merge conflicts. 

I think the best solution would be to add a default keymapping file and a custom key mapping file, so your users can easily merge the firmware update. 

Also, please understand that I don't want to spend 2 days understanding what goes on in the code. I bought this keyboard for 649 and I expect it to work, and don't have Bluetooth issues or needed maintenance code work. It is a keyboard, not an arduino or so. :)

Here is the PR with the merge conflicts: https://github.com/Valentin-Seehausen/Adv360-Pro-ZMK/pull/3

Thanks!

ReFil, 2023-06-07:
[image]
here are the &none entries that need removing in adv360.keymap

Valentin-Seehausen, 2023-06-08:
Ok so I basically accept all your diffs and none of mine?

Thanks for the prompt reply. Have to say support is amazing.

Valentin-Seehausen, 2023-06-10:
@ReFil ?

ReFil, 2023-06-10:
Apologies, yes if you accept my changes the merge conflicts will be resolved however your custom keymap will not be reserved. The easiest way is probably to screenshot it in the editor and recreate it

Valentin-Seehausen, 2023-06-12:
Come on, that is not a solution! This is a high-end product. 

What about the solution I proposed?

ReFil, 2023-06-12:
An alternative option is removing the unnecessary &none entries in your keymap as i detailed [here](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/167#issuecomment-1580961867)

gridlocdev, 2023-06-13:
I've put together some ideas in Discussion #170 for how these types of upgrades could be made a bit smoother, since I ran into this as well.

Valentin-Seehausen, 2023-06-13:
> The update removes 8 "&none" entries from each layer in the adv360.keymap and keymap.json, choosing your own diff will result in keys being shifted across.
> 
> you can see in the diff where the entries are removed, just remove them from your own keymap in each layer and it will work https://github.com/Valentin-Seehausen/Adv360-Pro-ZMK/pull/3/files#diff-df49145a9e31146b8135f11531a1734a3ec03e1340e9eb6a9016cbb39aea7308

For the records, I removed the &none entries from my file and I now have these merge conflicts: https://github.com/Valentin-Seehausen/Adv360-Pro-ZMK/pull/4

ReFil, 2023-07-05:
Closing as solved with the addition of instructions to resolve the conflict

closed 2023-07-05 by ReFil