KinesisCorporation/Adv360-Pro-ZMK #172: Bluetooth keeps reconnecting even after manual disconnection

cvcore, 2023-06-16:
Hi, 

When using the ADV360PRO keyboard on my MacBook, I occasionally need to take the laptop to other rooms and in this case, I often observe a very high latency when typing on the internal keyboard. This issue can be resolved after powering off the ADV360 before leaving the room. So I believe the high latency was caused by the poor bluetooth connection between the keyboard and laptop.

Due to this reason, I need a easy way to disconnect the keyboard. However, I'm simply not able to disconnect it from macOS's bluetooth dropdown menu. After clicking the Adv360 Pro icon for disconnection, after 1 second it automatically reconnects itself, which is quite annoying when you cannot leave the room to turn off the keyboard physically. Switching the keyboard to other profiles also doesn't help, since the keyboard always attempt to connect to all paired devices.

So with this github issue, I would like to request the following fixes:

- Ability to disconnect the keyboard from bluetooth host, without having to disable bluetooth completely
- Option to just connect to one host at a time from the keyboard

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/assets/4695324/49f1e822-a58a-4f22-a864-3252f583bd25

ReFil, 2023-06-19:
This is the standard behaviour of ZMK, but i can understand why it would present an issue in your use case. I will investigate the code to see if this can be changed

cvcore, 2023-07-07:
@ReFil gentle bump on this topic ;) 
Hopefully this weekend I can find some time to also look into the zmk firmware. Could you also give me some pointers to start with?

ReFil, 2023-07-07:
The adv360 uses a fork of ZMK stored here https://github.com/ReFil/zmk/tree/adv360-z3.2
For development you can fork my repo and change your config repos west.yml file to point to your fork.

The underlying zephyr RTOS has absolutely excellent documentation but remember to select the right zephyr version as ZMK lags the latest available version (there's a drop down menu on the left side)

All the relevant bluetooth stuff is in app/SRC/ble.c
You might be able to make it disconnect when it changes profile but I'm not sure how best to do that, I haven't given it much thought

Sorry I haven't had the time to look into it as I've been very busy with lots of work.

cvcore, 2023-07-11:
@ReFil Thanks for the pointers and from `app/SRC/ble.c` I believe there is something we can do about the advertising. The current logic will keep broadcasting when the current profile is open or loses connection. I believe that's probably something making the host computer keep reconnecting to the keyboard.

Maybe we can check the standby status of the keyboard and disable BLE advertising when it's in sleep mode (when the LEDs are off). In that way we can save some batteries, too.

Which function is best suited for reading the current standby status of the keyboard?

ReFil, 2023-07-22:
You'd do that by using the events system in zephyr. If you look at the end of rgb_underglow.c you can see how the Auto off on idle mechanic works. Anything conditional on CONFIG_ZMK_RGB_UNDERGLOW_AUTO_OFF_IDLE takes input from the sleep state of the board

cvcore, 2023-07-23:
Thanks for the reply @ReFil and today I got some time to look deeper into the code. So far I was able to implement the desired behavior with some minor logic changes of the code. Would be great if you could look into it and IMO it's better to make it as a standard behavior.

ReFil, 2023-07-23:
I'm not sure about introducing a drastic change in the Bluetooth behaviour as default. It would be better if it were to be behind a kconfig option you can enable in adv360_left_defconfig

KinesisCorporation, 2023-10-20:
@cvcore Please try again with today's update. We tested this exact scenario and didn't have any problems.

cvcore, 2023-10-27:
> @cvcore Please try again with today's update. We tested this exact scenario and didn't have any problems.

Latest update didn't work well for me. Please refer to: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/198#issuecomment-1783545633

And there is still no bluetooth disconnection handling implemented in the latest firmware:

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/assets/4695324/465cb683-6cd6-40e6-941f-9e483814252a

ReFil, 2023-11-13:
There are a few upstream ZMK PRs that stand to implement a behaviour to explicitly disconnect and handle disconnections better

ReFil, 2023-11-21:
https://github.com/zmkfirmware/zmk/pull/1638 This now merged PR should perform the functionality you want, in the next minor 360 pro update it will be brought into the 360 pro

ReFil, 2024-01-15:
Please try recreating the problem now, there are also new bluetooth behaviours to allow you to disconnect from a host with a keypress

cvcore, 2024-01-16:
Hi @ReFil , happy new year and great to hear the new features are here! I will try to update my kb to the new fw version on the upcoming weekend and let you know test the results.

cvcore, 2024-01-21:
Hi @ReFil ,

I'm able to recompile and flash the latest firmware with `BT_DISC` behavior implemented (https://github.com/zmkfirmware/zmk/pull/1638), however it didn't solve the issue for me after some testing. 

From my understanding, the PR implements the functionality which allows you to disconnect a profile from the keyboard, but not from the computer. To make it work, you need to have a second profile **paired** and **connected**, select this profile and only then it's possible to disconnect the first profile by hitting the key `BT_DISC`. Otherwise, the keyboard will still try to reconnect itself forever.

To me, this behavior is not useful given I need to disconnect the BT connection when I'm away from the keyboard. Additionally it seems for me the repeating key issue is back from this version.

zmk: https://github.com/cvcore/zmk/commit/e6a9407fea88c08bea97fad136ed9c16bd01acc1
adv360-pro-zmk: https://github.com/cvcore/Adv360-Pro-ZMK/commit/24a6b17f5465a21a21d8f692958c0101f5d7730c

ReFil, 2024-01-21:
For the repeating key issues try this fix https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/349#issuecomment-1901919337

cvcore, 2024-01-27:
> For the repeating key issues try this fix [#349 (comment)](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/349#issuecomment-1901919337)

Thanks for the hint. Actually I took another approach by implementing `bt_conn_disconnect` calls during profile switch. In addition switched back to `ADV_DIR` instead of `ADV_CONN` for advertising to paired devices. https://github.com/cvcore/zmk/compare/b281cb4b7230d4fcb86bc69ba85bb17819b74ed5...8996f95868134cdfe35ae6ae047ba5b2cb377497

It works for MacBooks, but far from perfect as now iPad can't recognize `ADV_DIR` broadcasts. For that I have to switch to an open profile to force `ADV_CONN` and then switch to the profile paired with iPad.

closed 2024-01-27 by cvcore

ReFil, 2024-01-28:
That's a neat solution, it's certainly an interesting implementation. It might be worth submitting this feedback as a feature request to upstream ZMK main. 

The reason ADV_DIR doesn't work is because apple don't permit it, as per section 49.3 of the (Accessory device specifications)[https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf], Directed advertising isn't technically permissible for connecting to apple devices