KinesisCorporation/Adv360-Pro-ZMK #179: Switching mod to next device doesn't work

venkatasreekanth, 2023-07-05:
When I try to clear an mod+2 and pair a new device the device goes back mod+1 device though it shows blue color

ReFil, 2023-07-05:
It will show as connected to any paired device but will only output to the selected device

venkatasreekanth, 2023-07-05:
The problem is select mod +2 but it sends output to mod +1 while showing
the color of mod + 2

ReFil, 2023-07-06:
Does this happen with all the Bluetooth profiles? Have you tried another?

venkatasreekanth, 2023-07-06:
Yes, tried with Mod +3, same issue.
Regards,

Venkata Sreekanth Bhagavatula

KinesisCorporation, 2023-07-06:
Is this an issue with a new keyboard, or had it been working fine until a firmware update?

venkatasreekanth, 2023-07-06:
The keyboard came with an older firmware, I upgraded it to version 2 with a
slight modification in keymap for mac. Otherwise nothing else changed. I
updated the firmware a couple of months ago.
Regards,

Venkata Sreekanth Bhagavatula

KinesisCorporation, 2023-07-06:
Understood. Did switching Profiles work for some period of time that after keymap update a couple of months ago?

venkatasreekanth, 2023-07-06:
I don't recall, but at some point I did switch between computers but I am
not sure if it was before or after the keymap/firmware update. I am in need
to switch between computers and devices now though.

Regards,

Venkata Sreekanth Bhagavatula

KinesisCorporation, 2023-07-06:
Just wondering if perhaps the BT Select commands may have been scrambled. If they haven't. It's probably best to perform a Settings Reset and install factory firmware to see if its a hardware defect. 

[image]

For more personalized support, and if necessary a Return Merchandise Authorization, please submit a trouble ticket here:
https://kinesis-ergo.com/support/contact-a-technician/

venkatasreekanth, 2023-07-06:
Do you want me to install version 2 or version 3? when you say factory
reset, does it mean no keymap?
Regards,

Venkata Sreekanth Bhagavatula

KinesisCorporation, 2023-07-06:
Either version 2.0 or 3.0 should be sufficient to test for this issue. Yes, factory version means the default layout so you would lose your customizations. But it should be a simple matter to test and rebuild the f/w if necessary.

venkatasreekanth, 2023-07-06:
I will try it after work and let you know.

Regards,

Venkata Sreekanth Bhagavatula

ReFil, 2023-07-25:
Hi, following up on this issue, was the problem solved with the factory firmware?

ReFil, 2023-07-29:
Just following up on this, have you been able to solve this issue?

ReFil, 2023-07-31:
Hi, just following up one last time, if i dont hear back in 48h i'll close out this issue as solved

closed 2023-08-02 by ReFil