KinesisCorporation/Adv360-Pro-ZMK #182: Hotkeys not working

larry-cdn77, 2023-07-06:
Hi, I get no response from the hotkeys 1-4 after upgrading to firmware version 3.0 from commit 771d68f

Is there a 'stable' version 3.0 I should be using?

Could something else be the reason I seemed to have lost the hotkeys?

KinesisCorporation, 2023-07-06:
The hotkeys have no default action in the production build.

At the factory we install a special QC version that assigns keypad 1-4 for testing.

You should be able to customize them as you see fit or ignore them.

larry-cdn77, 2023-07-08:
Thanks for the reply. I was able to customise by editing adv360.keymap (started off with the wrong guess that the file to edit is keymap.json). Used consumer codes yellow, red, green and blue (it helped to refer to https://zmk.dev/docs/codes).

The keys did something out of the box and stopped doing something after a version 3 firmware upgrade. That's what tripped me.

GitHub Actions CI worked a treat and so did the upgrade procedure (using Mod+HK1 and Mod+HK3 to put modules into bootloader mode)!

closed 2023-07-08 by larry-cdn77