KinesisCorporation/Adv360-Pro-ZMK #183: GUI Link Broken?

JunjieLeiCoe, 2023-07-08:
> https://kinesiscorporation.github.io/Adv360-Pro-GUI  

Is the GUI version link broken?

ReFil, 2023-07-08:
That link seems to work for me, what issue are you encountering?

JunjieLeiCoe, 2023-07-08:
The error message was related to `Unsecure Connections`, which I think might be related by the Chrome SSL certifications.

ReFil, 2023-07-09:
Which version of chrome and which operating system? I am unable to replicate this issue with several versions of chrome on android and linux

leonnardo, 2023-07-12:
Same here
Chrome 114.0.5735.199 on Windows 11

JunjieLeiCoe, 2023-07-17:
```
ERR_SSL_PROTOCOL_ERROR

```
Version 114.0.5735.199 (Official Build) (64-bit)

ashafaei, 2023-07-21:
I had the same issue. turns out the URL is blocked by my firewall. if you visit the HTTP version, you'd know if it is being intercepted or not.

ashafaei, 2023-07-25:
the issue is not the SSL certificate. for some reason, the firewall doesn't allow access to the URL and tries to intercept the connection. But since it can't/won't impersonate the destination, we end up getting SSL errors.

ashafaei, 2023-07-25:
when I access the url on `http` so that the interception works, this is what I get 
[image]
the question is why is it being blacklisted. the solution is changing your network or tunnelling your traffic through a node outside your network, or just disable any firewall in the middle temporarily.

ReFil, 2023-07-25:
That indicates it's a firewall on your ASUS router that's causing the problem, not anything on the remote side. Have you tried accessing the site using 4g on your phone?

ashafaei, 2023-07-25:
I already sorted it out. I'm just explaining why some people might experience this problem. Asus is outsourcing their blacklist to TrendMicro, https://global.sitesafety.trendmicro.com/result.php and if they have automatically blacklisted the page, the chances are that other firewalls do it too. the firewalls try to intercept your connection, and then the browser complains about the certificate.

ReFil, 2023-07-25:
Ok, thanks for the explanation. I'm going to close the issue as solved l

closed 2023-07-25 by ReFil