KinesisCorporation/Adv360-Pro-ZMK #185: Update to V3 disables right half

eckig, 2023-07-12:
I updated my repo to the V3.0 branch but my right half does no longer connect to the left (Status LED flashes red 1 time on the right side).

I tried all the steps to mitigate this:

1. Settings Reset file (multiple times).
2. Power cycling (multiple times).
3. Paper clip to tap both the left and right reset button in rapid succession.
4. Connect both modules to power.

Only thing that helped: Settings Reset (again) and then using my previous (V2.0) config.

ReFil, 2023-07-12:
Did you update both sides? If the red led flashes once then extinguishes that usually means it connected. Have you tried selecting a layer to see if the leds on the right side update?

ReFil, 2023-07-12:
Please note you have to use the settings reset file for V3.0 when updating

eckig, 2023-07-13:
1. Yes, I updated both sides. And no: Changing the layer did nothing to resolve this, also tried to connect via USB cable.
2. I used the `settings-reset.uf2` file from the V3.0 branch (root directory of this repo).

ReFil, 2023-07-13:
If you scan for Bluetooth devices is a device called "Adv 360 pro rt" visible?

eckig, 2023-07-13:
I didnt check this, because I need my keyboard to work and thus I reverted everything.

But I can tell, that the left half was working perfectly fine with the new firmware (at least connected with a cable).

If you need to know this, I will try it again on the weekend and check if the keyboard is visible via bluetooth.

vlad-psh, 2023-07-14:
Same here.

I spend thirty minutes resetting both sides simultaneously/one-by-one with a paperclip with no avail.

> If you scan for Bluetooth devices is a device called "Adv 360 pro rt" visible?

It was visible. I also tried unpairing and pairing it again. Nothing worked for me.

Left side was working (with a 50% chance after each paperclip "reboot"), but right side never connected.

Rolled back to the v2.0 and got it working again.

ReFil, 2023-07-14:
Please try the 3.0 factory default firmware and 3.0 settings reset available here https://kinesis-ergo.com/support/kb360pro/#firmware-updates

vlad-psh, 2023-07-14:
> Please try the 3.0 factory default firmware and 3.0 settings reset available here https://kinesis-ergo.com/support/kb360pro/#firmware-updates

Thanks for reply!

This was indeed the reason of the problem in my case. After flashing 'settings-reset-v3.uf2' my custom config from v3.0 branch started to work 👍

ashafaei, 2023-07-20:
I went through the same and the key was to flash with the `reset` file first, then flash with my custom firmware. I also put the keyboard into bootloader mode with the hidden reset button instead of the Mod keys. Mod keys didn't work on the right keyboard the first time I tried. Now if only there was a firmware to relieve me of this pain I'm going through to relearn typing.

eckig, 2023-07-22:
> Please try the 3.0 factory default firmware and 3.0 settings reset available here https://kinesis-ergo.com/support/kb360pro/#firmware-updates

That worked for me aswell, thanks for the help!

Maybe it would be a good idea to put the correct settings-reset file on the V3.0 branch?

ReFil, 2023-07-23:
I thought it was updated but will double check, thank you for the feedback, closing as solved

closed 2023-07-23 by ReFil