KinesisCorporation/Adv360-Pro-ZMK #192: How to configure `&ht_.*` or change `&mt { mod-term-ms }`

mjacobus, 2023-07-24:
Hi there

I just received my advantage 360 pro, and I am trying to make a smooth transition from my [erdgodox ez.](https://configure.zsa.io/ergodox-ez/layouts/lA6VG/xDvDp/0)

I found a few blockers:

Apparently, there is no way of choosing hold-tap behaviors in the advantage 360 [GUI](https://kinesiscorporation.github.io/Adv360-Pro-GUI/).

So the next best thing I could do, apparently, is to change the tapping-term-ms, as described here - https://zmk.dev/docs/behaviors/mod-tap#configuration

However, if I add those 3 lines before the keymap section, the build will fail. And if I were to edit the main config file, it would be rewritten by the GUI commit.

I noticed, there is a `cust_behaviors.json` file, but I did not find any documentation explaining how to use it.

So my questions are 3:

1 - How can I set hold-tap behaviors?
2 - How can I change the `tapping-term-ms`?
3 - How can I change custom configs, as \#2, and not have them overwritten by the GUI commits?

Thank you in advance.

ReFil, 2023-07-24:
The adv360 repo is intercompatible with nick coutsos' GUI, you should be able to configure your advanced behaviours using that GUI https://nickcoutsos.github.io/keymap-editor/

mjacobus, 2023-07-25:
@ReFil 

Thank you so much! That GUI is much better, IMO.

I think in the adv360 repo

ReFil, 2023-07-25:
No worries, glad to hear you got a solution working. Closing the issue as solved

closed 2023-07-25 by ReFil