KinesisCorporation/Adv360-Pro-ZMK #198: Serious Bluetooth regression in version 3

yanshay, 2023-07-28:
I've been using version 2 for several months now, using bluetooth on both Mac and Windows and bluetooth worked pretty much ok, some issues from time to time but nothing major.

I just upgraded to version 3 here, and have serious bluetooth issues, managed to get it working so that it barely work on the Mac but not at all on windows. 
Currently in order to connect to the Mac, every time I turn on the keyboard I have to remove all devices on the Mac and then repair. It appears that every time the keyboard starts up it has a new id, so a new keyboard appears on the bluetooth preferences list, so I have to force removing all devices (just the adv360 is not enough) and reconnect.
On windows it just won't connect, the operation fails immediately.
This makes this version totally unusable.

BTW - in order to get it working in the first place I had to reset the keyboard both with the settings that clear the flash and the reset-settings for version 3 (from the bottom of the page at https://kinesis-ergo.com/support/kb360pro/)

I'm not very familiar with zmk code base, but I think I've seen there were changes made to the ble module that haven't been merged to the code base used here, maybe that's the cause for these issues.

ReFil, 2023-07-28:
All changes from ZMK were merged in when the V3 update was performed. 
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/2d369451ae2a1326f64c37221802e1f4079b590a/config/boards/arm/adv360/adv360_left_defconfig#L52-L57
These lines alter the bluetooth configuration with the goal of improving the bluetooth experience for some mac users who were experiencing issues. Looking at your config repo you have some very odd configuration things going on, have you tried using the factory V3 files to see if there's any change in behaviour?

ReFil, 2023-07-28:
https://github.com/yanshay/Adv360-Pro-ZMK/blob/main/config/west.yml Your west.yml references a version of V2 firmware, not a version of v3 firmware. V3 firmware is derived from this branch https://github.com/ReFil/zmk/tree/adv360-z3.2

yanshay, 2023-07-28:
> https://github.com/yanshay/Adv360-Pro-ZMK/blob/main/config/west.yml Your west.yml references a version of V2 firmware, not a version of v3 firmware. V3 firmware is derived from this branch https://github.com/ReFil/zmk/tree/adv360-z3.2

I'm using a local repo clone, not the one on Github before I switch to it, so what you see on Github is not the one I'm using. I use podman locally to build and it points to your z3.2 branch.

yanshay, 2023-07-28:
> All changes from ZMK were merged in when the V3 update was performed. [...]

I'm referring to changes in the file `/app/src/ble.c` - current file on zmk main differs from the one in your branch. As I said, I don't know the codebase and only suspect the ble.c has to do with bluetooth.

ReFil, 2023-07-28:
There have been changes to ZMK main ble.c since the V3 update, at the time of the update they were the same. Please try the factory V3 firmware available here: https://kinesis-ergo.com/download/advantage360-professional-factory-default-7-6-23/ and the V3 settings reset available here: https://kinesis-ergo.com/download/advantage360-professional-factory-default-7-6-23/

yanshay, 2023-07-28:
> There have been changes to ZMK main ble.c since the V3 update, at the time of the update they were the same. Please try the factory V3 firmware available here: [...]

You provided the same link twice, can you provide the correct setting reset link to make sure I try the proper one?

ReFil, 2023-07-28:
My apologies, here is the link https://kinesis-ergo.com/download/adv360-pro-settings-reset-version-3-0/

yanshay, 2023-07-28:
I tried with the above files, I reset both sides, then installed the firmware and same behavior. In short - exact same behavior. 
After I remove all devices on the Mac I can connect the keyboard and it works.
Then I turn keyboard off and turn on, it blinks slowly and doesn't connect. It appears on the bluetooth settings of the Mac as a new device to connect to but fails to connect when pressing the button to connect. To get it connected I have to remove all devices on the mac (just this one is not enough, need to use the advanced remove all devices) and need to press the BL reset keyboard sequence to get it blinking fast. Only then it connect again.
I can see the hex 12 digits code on the Mac of the devices, every time it turns on and off (both if previously it was successfully connected or not) there is a new code. My other bluetooth keyboard keeps the same code all the time no matter what I do so I guess that's causing the issue, it just doesn't persist the code across sessions.

And on my windows machine it just doesn't connect, windows sees it and when I press to connect it quickly respond with 'try connecting your device again'

All these issues don't happen with the the v2 version (well, at least didn't until I tried to upgrade), will need to switch back soon and will know for sure.

ReFil, 2023-07-28:
The code changing behaviour you're describing is Bluetooth LE privacy. This is a feature apple explicitly requires in their accessory device specifications. It can be disabled by removing the lines I previously mentioned from adv360_left_defconfig. 
During testing of v3.0 i was unable to replicate the behaviour you describe on windows or Mac, what computers and Macos version are you using?

yanshay, 2023-07-28:
Is this a new requirement from apple? The other keyboard I mentioned that doesn't change its code is an apple magic keyboard.
I also don't see these lines in the 2.0 configuration file, so maybe that's indeed the issue.
Do I need to remove all lines? Or just the privacy one change to 'n'?

I'm using MacBook Pro 2018, still with BigSur, and Windows 10 on my PC.

ReFil, 2023-07-28:
It's been a requirement recently, although i do not know if it tracks back to big sur, remove only the privacy line and try again

major updates will often change the encryption keys on the keyboard, for the windows issue are you trying to pair to the keyboard or connect? you need to remove the pairing in windows and re-pair

yanshay, 2023-07-28:
I removed the `CONFIG_BT_PRIVACY=y ` line and now bluetooth works properly. So probably this doesn't work well with BigSur.

As for the windows machine, I don't see there the advantage keyboard, what I do is 'add' device, it scans and shows me devices and the pairing doesn't work. I restarted the machine to see there are no leftovers in memory. It's an external bluetooth dongle that worked with version 2 but don't know what vendor exactly.

Could it be one of the other lines?

ReFil, 2023-07-28:
unlikely but you can try. the other 5 lines work as one block, you have to remov them all.

yanshay, 2023-07-29:
I had some unstable connectivity issues with the Mac, some key-presses were delayed, some just didn't arrive. I removed those 5 other lines and things improved considerably. I still see cases where there is a miss but rare enough to think it might be me and not the bluetooth connectivity. Is there anywhere that documents those settings for non BT experts? couldn't find a clear explanation anywhere.

Also, it doesn't work with my windows machine. Any way on windows to verify it doesn't have any memory of old bluetooth pairing somewhere?

ReFil, 2023-07-29:
What Bluetooth adapter are you using on your windows pc? Make sure it is running the latest drivers

yanshay, 2023-07-31:
The bluetooth adapter on my Windows is called 'soomfon', it's using Realtek drivers. I updated the drivers and it still doesn't connect. Any ideas what I can check?

As for the Mac, from time to time the bluetooth connectivity becomes very poor. Not as regular as with those 5 lines above, but it still happens and then I have to connect to USB to be able to work. I think (though not sure) it's happening after some time of not using the keyboard, maybe due to it going to sleep?

Would it help to play with those BT settings in those 5 lines I removed? I found various references on the web to those parameters in other contexts with very different values, but don't understand what they mean.

ReFil, 2023-07-31:
The top line simply enables overriding of the settings, the next 4 are the actual parameters. We used the apple values available on page 193 of this document https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf
I dont know where you'd start with changing those values as they have a huge range but could severely degrade the experience if set incorrectly. What sort of connectivity issues are you encountering? is it lockops?

yanshay, 2023-07-31:
I experience mostly:
1. Delays - I press a key and it takes time until it takes effect, sometime up to what seems like 1/2 a second. And not a single key, when that happens it happens quite frequently to the point I can't work with the keyboard.
2. Events don't arrive, sometimes the key, sometimes the release of the key (and then it autorepeats forever). Also, when it happens it happens many times, not just once.
3. Bluetooth connects/disconnects - I see that when I keep the bluetooth preferences window open and I see the keyboard flips between connected/disconnected. If I wait (could be 10-15 seconds) it usually ends up back to be connected.

When it happens I connect to USB (I can do that quickly) and that behavior disappears. So it doesn't like like keyboard SW or laptop hickups, feels like connectivity. And on USB it never happens.

And other bluetooth devices at the same time are completely stable and function perfectly.

yanshay, 2023-07-31:
> The top line simply enables overriding of the settings, the next 4 are the actual parameters. We used the apple values available on page 193 of this document https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

This is dated 2022, is there such a document from earlier years? My laptop is 2018, maybe they changed chipsets/sw and there were other values for older HW? Like we've noticed that that privacy parameter completely failed the connection probably because my OS/Laptop are older?

yanshay, 2023-07-31:
```
CONFIG_BT_PERIPHERAL_PREF_TIMEOUT=600 
 CONFIG_BT_PERIPHERAL_PREF_MIN_INT=12 
 CONFIG_BT_PERIPHERAL_PREF_MAX_INT=24 
 CONFIG_BT_PERIPHERAL_PREF_LATENCY=30 
```
From the document:

> Peripheral Latency of up to 30 connection intervals

Seems like above setting is at the max latency allowed

> Supervision Timeout from 2 seconds to 6 seconds.

Is that the 600 above? what units are those?

> Interval Min of at least 15 ms.

Isn't the relevant number above 12 and it's less than the minimum?

> Interval Min is a multiple of 15 ms.

And it isn't a multiple of 15?

> Interval Max * (Peripheral Latency + 1) of 2 seconds or less.
> Supervision Timeout greater than Interval Max * (Peripheral Latency + 1) * 3.

?

ReFil, 2023-08-04:
The delay and lockup issue could also be connected to this fix https://github.com/zmkfirmware/zmk/commit/9a963abfc8b27c6fdfb5b6b4b38e5905d6a360a7 which has yet to be merged into the adv360 branch but is on the list of things to do

jasselin, 2023-08-06:
I have the same behavior here with repeating keys, even when I use my Macbook keyboard while the 360 is connected. No repeat or delay when I shut it off. Any idea when this could be pulled from the zmk repo?

ReFil, 2023-08-09:
There's now a beta firmware with that change merged. It would be greatly appreciated if yyou could try it and give feedback https://kinesis-ergo.com/support/kb360pro/#firmware-updates

cvcore, 2023-08-10:
> There's now a beta firmware with that change merged. It would be greatly appreciated if yyou could try it and give feedback https://kinesis-ergo.com/support/kb360pro/#firmware-updates

Do you have a link to your branch with that commit merged?

cvcore, 2023-08-10:
> I have the same behavior here with repeating keys, even when I use my Macbook keyboard while the 360 is connected. No repeat or delay when I shut it off. Any idea when this could be pulled from the zmk repo?

@jasselin my solution atm is just to implement a feature that allows you disconnect the keyboard from macbook's bluetooth menu when that happens. Not perfect, but works for me in most cases.
Ref: https://github.com/ReFil/zmk/pull/7

ReFil, 2023-08-14:
> > There's now a beta firmware with that change merged. It would be greatly appreciated if yyou could try it and give feedback https://kinesis-ergo.com/support/kb360pro/#firmware-updates
> 
> @ReFil Do you have a link to your branch with that commit merged?

Sorry I dont as of yet, it's just a test branch on my laptop. I can push it if needs be

cvcore, 2023-08-14:
> Sorry I dont as of yet, it's just a test branch on my laptop. I can push it if needs be

That would be great, so I can also test it with my own configurations. Thanks!

ReFil, 2023-08-15:
https://github.com/zmkfirmware/zmk/issues/1900 This could possibly be connected

yanshay, 2023-08-17:
> That would be great, so I can also test it with my own configurations. Thanks!

Same here. 
BTW - I witness more and more BT related issues making the keyboard unusable with BT for long time.

cvcore, 2023-08-23:
> Sorry I dont as of yet, it's just a test branch on my laptop. I can push it if needs be

@ReFil any updates?

ReFil, 2023-08-24:
https://github.com/ReFil/zmk/tree/adv360-z3.2-beta

apologies for the delay

cvcore, 2023-08-26:
> https://github.com/ReFil/zmk/tree/adv360-z3.2-beta
> 
> apologies for the delay

I switched to this branch with my own keymaps modifications and also merged additional logics I made to handle the BLE disconnection from host computer. After around 1h of usage I don't observe the repeating keys issue as before. Also typing on the MacBook internal keyboard no longer gives me a laggy feeling anymore.

I will follow up when I observe any new issues from this version. But so far it's a great improvement for me. Many thanks for the work :)

KinesisCorporation, 2023-08-29:
Still working better?

cvcore, 2023-08-29:
> Still working better?

Yes :) Tested on my both MacBooks (2021, Intel and 2023, M2).

lyallcooper, 2023-09-04:
I experience this issue as well, although the main manifestation is when I take my laptop (2021 M1 MacBook Pro, macOS 13.5) away from my desk (and Adv360) and into another room. My laptop keyboard will become unusable due to dropped, repeated, laggy keys (as described above) unless I turn off bluetooth or the Adv360 itself.

@cvcore's patch helps in that it lets you disconnect the keyboard from the bluetooth menu without turning off bluetooth entirely, but neither it nor the updated version of zmk the branch is based on fix the underlying issue. If anything the performance seems a little worse on the [`ReFil:adv360-z3.2-beta`](https://github.com/ReFil/zmk/tree/adv360-z3.2-beta) branch that it does the stable `adv360-z3.2` one? But thank you to both @cvcore and @ReFil for working on this!

In general, the bluetooth performance for my Adv360 has been significantly worse that the other keyboard I have that also runs zmk, and I (poorly) soldered that one together myself so if anything it should be the other way around. It's not bad enough to make the Adv360 unusable, but it's annoying. And the difference between using the Adv360 over bluetooth vs over USB with bluetooth disabled is like night and day.

KinesisCorporation, 2023-10-20:
> I experience this issue as well, although the main manifestation is when I take my laptop (2021 M1 MacBook Pro, macOS 13.5) away from my desk (and Adv360) and into another room. My laptop keyboard will become unusable due to dropped, repeated, laggy keys (as described above) unless I turn off bluetooth or the Adv360 itself.
> 
> @cvcore's patch helps in that it lets you disconnect the keyboard from the bluetooth menu without turning off bluetooth entirely, but neither it nor the updated version of zmk the branch is based on fix the underlying issue. If anything the performance seems a little worse on the [`ReFil:adv360-z3.2-beta`](https://github.com/ReFil/zmk/tree/adv360-z3.2-beta) branch that it does the stable `adv360-z3.2` one? But thank you to both @cvcore and @ReFil for working on this!
> 
> In general, the bluetooth performance for my Adv360 has been significantly worse that the other keyboard I have that also runs zmk, and I (poorly) soldered that one together myself so if anything it should be the other way around. The bluetooth performance is not bad enough to make the Adv360 unusable for me, but it's quite annoying sometimes. And the difference between using the Adv360 over bluetooth vs over USB with bluetooth disabled is like night and day.

Please re-flash with today's release, test again and report back. We have tested physically separating the Adv360 from a 2023 MacBook Pro running 13.5.1 by walking it far enough away that the Mac and 360 become disconnected. The laptop keyboard works fine while the separated and re-uniting the keyboard Mac has the 360 keyboard working fine as well.

eugene8080, 2023-10-27:
I'm also having trouble with repeating keys after upgrading to v3.0. I did not have the issue on v2.0. I'm running a Surface Laptop Studio on Win11. I tried using the changed my west.yml file to read revision: adv360-z3.2-beta, but i keep getting a "Process completed with exit code 1." error after I commit changes and attempt a Build action.
what did I do wrong?
https://github.com/eugene8080/Adv360-Pro-ZMK/blob/V3.0/config/west.yml

ReFil, 2023-10-27:
Hi, theree was a bug in the online build process that has since been fixed, you should be able to build your firmwares now, apologies for the inconvenience

cvcore, 2023-10-27:
> Please re-flash with today's release, test again and report back. We have tested physically separating the Adv360 from a 2023 MacBook Pro running 13.5.1 by walking it far enough away that the Mac [...]

Hi @KinesisCorporation , today I got some time to test the v10-23-23 firmware downloaded from https://kinesis-ergo.com/support/kb360pro/#firmware-updates. I followed exactly the [instructions](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/13181176/Advantage360.Professional.-.Firmware.Update.Instructions.10.25.23.KB360-PRO-GBR.pdf) when flashing the firmware. Unfortunately, the test result is not satisfying.

I tested by typing two paragraphs of text from https://official-typing-test.com/test/10test.html and made [this screen recording](https://www.youtube.com/watch?v=1qvHdtb8AS4). The repeating keys issue happened for me 3 times within a time frame of 5 minutes (as can be seen in 0:54, 1:59, 3:04).

For a fair comparison, I've also used a HHKB-BT keyboard paired to the same macbook via Bluetooth and typed for 5 minutes on the same website. There were no repeating keys observed. So the Bluetooth connectivity shouldn't be an issue.

Some more hardware info to help debug:
- Laptop: MacBook Air M2, 2022 Model Mac14,2
- OS: macOS Ventura 13.0
- Bluetooth: Chip BCM_4387
- ADV360 has RSSI of -64 dBm when connected (in comparison my HHKB-BT has -45 dBm)

Since this issue is impacting many users already, I'm hoping for a systematic approach to identify the root cause of the issue before releasing a newer version. It's a great piece of hardware and I hope the software as well :)

eugene8080, 2023-10-28:
Is the ReFil:adv360-z3.2-beta still the best branch for bluetooth updates? I noticed the main branch now refers to ReFil:adv360-z3.2-2

KinesisCorporation, 2023-10-28:
@cvcore 

We should probably open new "issues" that are laser focused on one specific "bug". The latest update was targeting improvements in syncing between the left and right module, and stability of connection with the PC.

Please start a new issue pertaining to your repeating keys and we'll look to get to the root cause of that. Thanks!

yanshay, 2023-10-29:
> Please start a new issue pertaining to your repeating keys and we'll look to get to the root cause of that. Thanks!

When I had repeating keys issues those definitely were driven by BT issues. I've also had other symptoms. I'm not sure you want to fill this repo here with all BT symptoms one by one when they definitely feel like all driven by BT issues.

Anyway, the interference BT issues on my end seem to be much better since I opened this issue. The main thing I've done was to move Logitech wireless dongle further from my laptop and keyboard using a USB extension coord. I think the Advantage Pro is prone to BT interference from Logitech, maybe low transmittion or reception power. I saw someone else on this issue also wrote he had such issues. Other BT devices I have didn't have those issues, only the Advantage Pro had.

I also updated to the latest branch a few days ago, I think that improved on the disconnect issues which took place once in a while, I did not have that since I made the update.

ReFil, 2023-10-29:
Yes, the 3.2-2 branch uses the current latest updates. It implements the repeating key fix here: https://github.com/zmkfirmware/zmk/commit/9a963abfc8b27c6fdfb5b6b4b38e5905d6a360a7 as well as a host of other improvements. The 3.2-beta branch is currently equal with 3.2-2 however is subject to frequent, often breaking changes so unless you're willing to be on the bleeding edge it's recommended you use 3.2-2

eugene8080, 2023-11-01:
I thought the repeating keys issue went away after using adv360-z3.2-beta, but it came back again.
When I flashed the [v2.0 firmware](https://kinesis-ergo.com/wp-content/uploads/Adv360-Pro-Factory-Default-Firmware-v2.0-11-1-22.zip), I have no repeating keys issue at all.
Where can I find the old v2.0 files on Github so I can make changes to the keymap and use 2.0 until 3.0 becomes usable on my computer?

===============================================
Everything I have tried for v3.0 firmware has resulted in repeating keys.

I haved tried:
turning off my Logitech mouse (@yanshay)
disconnecting the logitech wifi dongle
turning off bluetooh on my phone and computer before turning on the adv360

For the Settings reset, i have used:
settings-reset.uf2 released 2023-10-30
settings-reset-v1.0.uf2, settings-reset-v2.0.uf2, settings-reset-v3.0.uf2 in sequence

Firmware I have now tested:
Adv360 Pro Factory Default Branch 3.0 (v10-23-23)
Adv360 Pro Factory Default Branch 3.0 (v7-6-23)
adv360-z3.2-2
adv360-z3.2-beta
===========================================
Keyboard: Kinesis Advantage360 Pro
Computer: Surface Laptop Studio, 11th Gen Intel(R) Core(TM) i7-11370H @ 3.30GHz
Bluetooth: Intel Wifi-6 AX200 160MHz
OS: Windows 11 Home 22H2

ReFil, 2023-11-06:
@yanshay just following up, have you encountered further issues since updating and making those setup changes?

yanshay, 2023-11-07:
> @yanshay just following up, have you encountered further issues since updating and making those setup changes?

It's much better and completely usable so far on BT.
I think I did notice a few cases where when turning on the keyboards (left then right) it doesn't work and the right side blinks error, but turning off and on again both gets it to work, so not a big deal for me. It then stays connected w/o noticeable issued.

ReFil, 2023-11-07:
For the left right connection try leaving the left on for a few seconds before turning the right on, since the 10/20/23 update having the left on for a bit (Once the leds come online it should be good) before turning the right on should help. If you're happy with the performance as is are you ok to close this issue out?

ReFil, 2023-11-10:
@cvcore please try this change that another macos user reported helped their situation significantly https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/262#issuecomment-1793689158

lodi, 2023-11-10:
Sorry to take this off topic but I found this thread while trying to troubleshoot why I couldn't connect via Bluetooth to an Android tablet.  The tablet would pair, but then wouldn't connect in any way.  I tried commenting out the `CONFIG_BT_PRIVACY=y` line as suggested above and it completely resolved the issue for me.

yanshay, 2023-11-11:
> For the left right connection try leaving the left on for a few seconds before turning the right on, since the 10/20/23 update having the left on for a bit (Once the leds come online it should be [...]

Yes, I'm ok with closing this issue. Thanks for resolving this.

ReFil, 2023-11-11:
@lodi In the newest update privacy has been disabled by default and so you shouldn't have any issues with it any more. Thanks for documenting your experience getting it working on Android

closed 2023-11-11 by ReFil